CANopenNode/CANopenNode #270: Dynamic assignement of RPDO and TPDO COB-IDs

WillyKaze, 2021-03-02:
Hi,

I'm trying to dynamically assign COB-IDs for T/RPDOs. PDO communication works fine when defined from OD but I need to set up those dynamically.

I'm using CANopenNode v1.3 running on a LPC1768.

Here is my use-case:

I have a CANopen device on the network that can have various node ID, depending of its config. The LPC1768 scans the network with SDO and find this device.
Unfortunately, this device doesn't support modifying its PDO IDs and they are calculated from its node ID, i.e.: `device node ID = 3`  will result in PDOs set to `0x203` and `0x183`.
This means I need to set the LPC1768 PDOs after scan so that they can talk to each other.

So far I tried:
- setting the LPC1768 OD directly using globals:
  ```cpp
  OD_RPDOCommunicationParameter[0].COB_IDUsedByRPDO |= 0x80000000;
  OD_RPDOCommunicationParameter[0].COB_IDUsedByRPDO = 0x80000183;
  OD_RPDOCommunicationParameter[0].COB_IDUsedByRPDO &= ~0x80000000;
  ```
  This doesn't work because there is some T/RPDO buffers that also need to be updated
- setting via SDO
  I get a `CO_SDO_AB_DEVICE_INCOMPAT` with every read/write attempt on `0x1600[1]` and `0x1800[1]`

Is there a way to do that dynamically? I tried to search for examples but couldn't find examples.
I also took a look at #261 and it seems I'm following the right logic, but not the right way.

Any advice is welcome :)

geoffrey-vl, 2021-03-02:
Yes it is working, I'm using it on linux when COMaster is preoperational. Finally, when everything is mapped, I manually set the master to operational.

First, for each PDO you have to setup the PDO Communication object and Mapping object.
So you need to lookup the PDO entry in the OD (which is not the PDO index!).
For example, following snippet does a lookup of a RPDO Communication object.
```
bool COPdoMapper::setupRPDOComm(int pdoNr, uint32_t cobid)
{
    const uint16_t ODIndex = 0x1400 + pdoNr;
    uint16_t entryNo = CO_OD_find(_sdo, ODIndex);
    if(entryNo >= NO_OD_OBJECTENTRY) {
        log_printf(LOG_ERR, "OD[0x%04X] RPDOComms is not an OD object", ODIndex);
        return false;
    }
    CO_OD_extension_t* ext = &(_sdo->ODExtensions[entryNo]);
    ...
```

Next you need to setup the RPDO Communication object:

```
    //update transmission type
    uint8_t subidx = 2;
    uint8_t transmissionType = 0xFF;
    log_printf(LOG_DEBUG, "OD[0x%04X] setting Transmissiontype (subindex %d) to 0x%02X", ODIndex, subidx, transmissionType);
    CO_SDO_abortCode_t code = setSubIndexU8(_sdo, entryNo, ext, ODIndex, subidx, transmissionType);
    if(code != CO_SDO_AB_NONE) {
        log_printf(LOG_ERR, "Failed setting RPDO comm subindex %d, errorcode 0x%08X", subidx, (uint32_t)code );
        return false;
    }

    //update COB-ID
    subidx = 1;
    log_printf(LOG_DEBUG, "OD[0x%04X] setting COBID (subindex %d) to 0x%04X", ODIndex, subidx, cobid);
    code = setSubIndexU32(_sdo, entryNo, ext, ODIndex, subidx, cobid);
    if(code != CO_SDO_AB_NONE) {
        log_printf(LOG_ERR, "Failed setting RPDO comm subindex %d, errorcode 0x%08X", subidx, (uint32_t)code );
        return false;
    }

    log_printf(LOG_DEBUG, "OD[0x%04X] setting MaxSubindex (subindex %d) to 0x%02X", ODIndex, 0, 0x02);
    CO_OD_RAM.RP
```

The setSubIndexU32 code looks as following:

```
inline CO_SDO_abortCode_t setSubIndexU8(CO_SDO_t* sdo, uint16_t entryNo, CO_OD_extension_t* ext, uint16_t index, uint8_t subIdx, uint8_t data)
{
    if( ext->pODFunc == NULL) {
        return CO_SDO_AB_UNSUPPORTED_ACCESS;
    }
    CO_ODF_arg_t odfArg;
    odfArg.object = ext->object;
    odfArg.reading = false; //we're writing to OD
    odfArg.subIndex = subIdx;
    odfArg.data = static_cast<uint8_t*>(CO_OD_getDataPointer(sdo, entryNo, subIdx));
    odfArg.attribute = CO_OD_getAttribute(sdo, entryNo, subIdx);
    odfArg.pFlags = CO_OD_getFlagsPointer(sdo, entryNo, subIdx);
    odfArg.dataLength = CO_OD_getLength(sdo, entryNo, subIdx);
    CO_setUint8(odfArg.data, data); //write to odfArg data buffer
    return ext->pODFunc(&odfArg); // functionpointer to either CO_ODF_RPDOcom, CO_ODF_RPDOmap, CO_ODF_TPDOcom, CO_ODF_TPDOmap
}
```

And so forth... Look for how SDO server applies the mapping, its likewise! I also found some problem with in what order things get setup. Always map first, and setup communication second for PDO to work! You may need to experiment a bit, it took me some time to get it  working.

CANopenNode, 2021-03-02:
You can also configure PDO dynamically in running CANopen device. You just have to follow the standard CANopen [PDO re-mapping procedure](http://can-newsletter.org/uploads/media/raw/e522172be523845be799af1e5d6dd1f7.pdf).

The point is, you have to disable the PDO first (SDO, set bit 31 in 0x1400,01). Then change the CAN-ID, mapping, etc, then enable the PDO by clearing bit31. (You have to write 0x1400,01 multiple times)

I have to make some tutorial on using PDOs.

WillyKaze, 2021-03-03:
Thank you both for your replies.

Today I tried to implement the setupRPDO functions but so far no luck, this is mostly related to the code running on a small platform. I need to investigate further on how to do this.

For the SDO method, this is what I tried to do originally:
```
- SDO read 0x1400[1] (4)                                // Get current COB-ID
- SDO write 0x1400[1] (4) (0x80000000 & current COB-ID) // set 31 bit to disable PDO
- SDO write 0x1400[2] (1) 254                           // set transmission type
- SDO write 0x1400[1] (4) (0x80000000 & new COB-ID)     // set new COB-ID
- SDO write 0x1400[1] (4) new COB-ID                    // clear 31 bit to enable PDO
```

I get `0x06040047` after the first SDO read. Is there a common cause for that SDO abort code?

WillyKaze, 2021-03-03:
This error was caused by the OD entry not having correct parameters. This works now with the SDO sequence described above, thank you for your help!

closed 2021-03-05 by WillyKaze