CANopenNode/CANopenNode #272: wrong sdo reply result, when sending rpdo at fast rate

npk48, 2021-03-12:
I'm having this very strange issue when reading sdo and sending sync pdo at the same time. 
Basically what I'm doing is sending rpdo at 2ms interval time to 0x4000 0x00 od. Meanwhile reading 0x4000 0x00 with sdo. 2 out of 10 times, it returns wrong reading. the data part of the sdo reply is exactly the same as the request i sent, with only cob-id difference.

it seems like inside the stack something is not properly lock protected.

CANopenNode, 2021-03-12:
Can you share some more information, which version and which target (driver) do you use.

npk48, 2021-03-13:
SHA-1: 41e982962e1be83ef13707e28b1221a07f5ffb40
* Merge branch 'odesenfans-common-header-for-drivers'
18/12/2019 
We are using this commit, I know it's quite old, but it works so we didn't update to latest version.

[https://gist.github.com/npk48/69e6a79acf0c0dfdf2741937db463e37](driver)
and the driver we are using for stm32f4

CANopenNode, 2021-03-13:
There is probably problem in the driver or with the main() function. Unfortunately I cant help with the driver or example for STM. Those drivers wasn't maintained, so they was removed from newer versions. Please check CO_LOCK_OD() macros. You can also look into CANopenPIC, PIC32 for reference.

npk48, 2021-03-15:
I have checked the driver
`c
#define CO_LOCK_OD()            __set_PRIMASK(1);
#define CO_UNLOCK_OD()          __set_PRIMASK(0);
`

the lock is properly defined by masking the global interrupt flag.

Maybe we need to lock this in the main while loop for sdo processing manually?

CANopenNode, 2021-03-15:
It is necessary to lock all functions in fast timer thread (CO_TPDO_process, etc) and each OD variable access from mainline. SDO server already have the lock.

I have recently documented better the locking mechanism. Please see here:
https://github.com/CANopenNode/CANopenNode/blob/fc817770974e31656eea89998f4f2c612432528a/301/CO_driver.h#L389

closed 2021-03-18 by CANopenNode