CANopenNode/CANopenNode #274: NewOD Branch compilation errors

ChSotiriou, 2021-03-14:
While trying to compile the newOD branch I get the following errors. I'm trying to port the code to an STM32F7. I copied and pasted the compilation errors from eclipse

```
Description	Resource	Path	Location	Type
'CO_EM_t' {aka 'struct <anonymous>'} has no member named 'inhibitEmTime_us'	CO_Emergency.c	/CANopenNode_stm32f7/CANopenNode/301	line 614	C/C++ Problem
'CO_EM_t' {aka 'struct <anonymous>'} has no member named 'inhibitEmTime_us'	CO_Emergency.c	/CANopenNode_stm32f7/CANopenNode/301	line 620	C/C++ Problem
'CO_EM_t' {aka 'struct <anonymous>'} has no member named 'inhibitEmTime_us'	CO_Emergency.c	/CANopenNode_stm32f7/CANopenNode/301	line 664	C/C++ Problem
'CO_EM_t' {aka 'struct <anonymous>'} has no member named 'inhibitEmTime_us'	CO_Emergency.c	/CANopenNode_stm32f7/CANopenNode/301	line 666	C/C++ Problem
'CO_EM_t' {aka 'struct <anonymous>'} has no member named 'inhibitEmTimer'	CO_Emergency.c	/CANopenNode_stm32f7/CANopenNode/301	line 614	C/C++ Problem
'CO_EM_t' {aka 'struct <anonymous>'} has no member named 'inhibitEmTimer'	CO_Emergency.c	/CANopenNode_stm32f7/CANopenNode/301	line 615	C/C++ Problem
'CO_EM_t' {aka 'struct <anonymous>'} has no member named 'inhibitEmTimer'	CO_Emergency.c	/CANopenNode_stm32f7/CANopenNode/301	line 620	C/C++ Problem
'CO_EM_t' {aka 'struct <anonymous>'} has no member named 'inhibitEmTimer'	CO_Emergency.c	/CANopenNode_stm32f7/CANopenNode/301	line 623	C/C++ Problem
'CO_EM_t' {aka 'struct <anonymous>'} has no member named 'inhibitEmTimer'	CO_Emergency.c	/CANopenNode_stm32f7/CANopenNode/301	line 664	C/C++ Problem
'CO_EM_t' {aka 'struct <anonymous>'} has no member named 'inhibitEmTimer'	CO_Emergency.c	/CANopenNode_stm32f7/CANopenNode/301	line 666	C/C++ Problem
'CO_LOCK_EMCY' undeclared (first use in this function)	CO_Emergency.c	/CANopenNode_stm32f7/CANopenNode/301	line 732	C/C++ Problem
'CO_ODF_arg_t' undeclared (first use in this function)	CO_SRDO.c	/CANopenNode_stm32f7/CANopenNode/304	line 682	C/C++ Problem
'CO_UNLOCK_EMCY' undeclared (first use in this function)	CO_Emergency.c	/CANopenNode_stm32f7/CANopenNode/301	line 756	C/C++ Problem
'ODF_arg' undeclared (first use in this function)	CO_SRDO.c	/CANopenNode_stm32f7/CANopenNode/304	line 795	C/C++ Problem
expected ';' before 'ODF_arg'	CO_SRDO.c	/CANopenNode_stm32f7/CANopenNode/304	line 794	C/C++ Problem
macro "CO_LOCK_EMCY" passed 1 arguments, but takes just 0	CO_Emergency.c	/CANopenNode_stm32f7/CANopenNode/301	line 732	C/C++ Problem
macro "CO_UNLOCK_EMCY" passed 1 arguments, but takes just 0	CO_Emergency.c	/CANopenNode_stm32f7/CANopenNode/301	line 756	C/C++ Problem
make: *** [CANopenNode/301/subdir.mk:52: CANopenNode/301/CO_Emergency.o] Error 1	CANopenNode_stm32f7		 	C/C++ Problem
make: *** [CANopenNode/304/subdir.mk:24: CANopenNode/304/CO_SRDO.o] Error 1	CANopenNode_stm32f7		 	C/C++ Problem
make: *** [Core/Src/subdir.mk:44: Core/Src/main.o] Error 1	CANopenNode_stm32f7		 	C/C++ Problem
make: *** Waiting for unfinished jobs....	CANopenNode_stm32f7		 	C/C++ Problem
request for member 'attribute' in something not a structure or union	CO_SRDO.c	/CANopenNode_stm32f7/CANopenNode/304	line 687	C/C++ Problem
request for member 'data' in something not a structure or union	CO_SRDO.c	/CANopenNode_stm32f7/CANopenNode/304	line 689	C/C++ Problem
request for member 'dataLength' in something not a structure or union	CO_SRDO.c	/CANopenNode_stm32f7/CANopenNode/304	line 690	C/C++ Problem
request for member 'index' in something not a structure or union	CO_SRDO.c	/CANopenNode_stm32f7/CANopenNode/304	line 684	C/C++ Problem
request for member 'object' in something not a structure or union	CO_SRDO.c	/CANopenNode_stm32f7/CANopenNode/304	line 686	C/C++ Problem
request for member 'object' in something not a structure or union	CO_SRDO.c	/CANopenNode_stm32f7/CANopenNode/304	line 799	C/C++ Problem
request for member 'ODExtensions' in something not a structure or union	CO_SRDO.c	/CANopenNode_stm32f7/CANopenNode/304	line 666	C/C++ Problem
request for member 'ODExtensions' in something not a structure or union	CO_SRDO.c	/CANopenNode_stm32f7/CANopenNode/304	line 678	C/C++ Problem
request for member 'ODExtensions' in something not a structure or union	CO_SRDO.c	/CANopenNode_stm32f7/CANopenNode/304	line 778	C/C++ Problem
request for member 'ODExtensions' in something not a structure or union	CO_SRDO.c	/CANopenNode_stm32f7/CANopenNode/304	line 791	C/C++ Problem
request for member 'pFlags' in something not a structure or union	CO_SRDO.c	/CANopenNode_stm32f7/CANopenNode/304	line 688	C/C++ Problem
request for member 'pODFunc' in something not a structure or union	CO_SRDO.c	/CANopenNode_stm32f7/CANopenNode/304	line 679	C/C++ Problem
request for member 'pODFunc' in something not a structure or union	CO_SRDO.c	/CANopenNode_stm32f7/CANopenNode/304	line 691	C/C++ Problem
request for member 'pODFunc' in something not a structure or union	CO_SRDO.c	/CANopenNode_stm32f7/CANopenNode/304	line 792	C/C++ Problem
request for member 'pODFunc' in something not a structure or union	CO_SRDO.c	/CANopenNode_stm32f7/CANopenNode/304	line 804	C/C++ Problem
request for member 'reading' in something not a structure or union	CO_SRDO.c	/CANopenNode_stm32f7/CANopenNode/304	line 683	C/C++ Problem
request for member 'subIndex' in something not a structure or union	CO_SRDO.c	/CANopenNode_stm32f7/CANopenNode/304	line 685	C/C++ Problem
unknown type name 'CO_OD_extension_t'; did you mean 'OD_extension_t'?	CO_SRDO.c	/CANopenNode_stm32f7/CANopenNode/304	line 678	C/C++ Problem
unknown type name 'CO_OD_extension_t'; did you mean 'OD_extension_t'?	CO_SRDO.c	/CANopenNode_stm32f7/CANopenNode/304	line 791	C/C++ Problem
unknown type name 'CO_ODF_arg_t'	CO_SRDO.c	/CANopenNode_stm32f7/CANopenNode/304	line 359	C/C++ Problem
unknown type name 'CO_ODF_arg_t'	CO_SRDO.c	/CANopenNode_stm32f7/CANopenNode/304	line 426	C/C++ Problem
unknown type name 'CO_ODF_arg_t'	CO_SRDO.c	/CANopenNode_stm32f7/CANopenNode/304	line 455	C/C++ Problem
unknown type name 'CO_ODF_arg_t'	CO_SRDO.c	/CANopenNode_stm32f7/CANopenNode/304	line 468	C/C++ Problem
unknown type name 'CO_ODF_arg_t'	CO_SRDO.c	/CANopenNode_stm32f7/CANopenNode/304	line 681	C/C++ Problem
unknown type name 'CO_SDO_t'	CO_PDO.h	/CANopenNode_stm32f7/CANopenNode/301	line 184	C/C++ Problem
unknown type name 'CO_SDO_t'	CO_PDO.h	/CANopenNode_stm32f7/CANopenNode/301	line 225	C/C++ Problem
unknown type name 'CO_SDO_t'	CO_SRDO.h	/CANopenNode_stm32f7/CANopenNode/304	line 126	C/C++ Problem
unknown type name 'CO_SDO_t'; did you mean 'CO_SRDO_t'?	CO_SRDO.c	/CANopenNode_stm32f7/CANopenNode/304	line 170	C/C++ Problem
unknown type name 'CO_SDO_t'; did you mean 'CO_SRDO_t'?	CO_SRDO.c	/CANopenNode_stm32f7/CANopenNode/304	line 483	C/C++ Problem
unknown type name 'CO_SDO_t'; did you mean 'CO_SRDO_t'?	CO_SRDO.c	/CANopenNode_stm32f7/CANopenNode/304	line 555	C/C++ Problem
unknown type name 'CO_SDO_t'; did you mean 'CO_SRDO_t'?	CO_SRDO.c	/CANopenNode_stm32f7/CANopenNode/304	line 669	C/C++ Problem
unknown type name 'CO_SDO_t'; did you mean 'CO_SRDO_t'?	CO_SRDO.c	/CANopenNode_stm32f7/CANopenNode/304	line 782	C/C++ Problem
unknown type name 'CO_SDO_t'; did you mean 'CO_SRDO_t'?	CO_SRDO.h	/CANopenNode_stm32f7/CANopenNode/304	line 178	C/C++ Problem
unknown type name 'CO_SDO_t'; did you mean 'CO_SRDO_t'?	CO_SRDO.h	/CANopenNode_stm32f7/CANopenNode/304	line 226	C/C++ Problem
unknown type name 'CO_SDO_t'; did you mean 'CO_TPDO_t'?	CO_PDO.h	/CANopenNode_stm32f7/CANopenNode/301	line 292	C/C++ Problem
unknown type name 'CO_SDO_t'; did you mean 'CO_TPDO_t'?	CO_PDO.h	/CANopenNode_stm32f7/CANopenNode/301	line 360	C/C++ Problem
```

CANopenNode, 2021-03-15:
PDO and SRDO should be disabled for now (PDO will be available shortly). Example is currently outdated, please check socketCAN/CO_driver_target.h for now.

use `#define CO_CONFIG_PDO (0)`

In CO_Emergency.c there were some missing statements "#if (CO_CONFIG_EM) & CO_CONFIG_EM_PROD_INHIBIT". I have just pushed fix to newOD.

CANopenNode, 2021-03-18:
Now PDO should work too. SRDO must still be disabled, but it is advanced feature, not necessary for usual CANopen device.

closed 2021-03-18 by CANopenNode