CANopenNode/CANopenNode #276: SDO Client usage

TorbjornG, 2021-03-18:
Hi!, have a project and we was looking at CANopenNode (current V2) , this system acts both as server and client, tried to find out how to use CO_SDOclient* and loading of different [OD] settings depending of server that the client shall communicate with.
Is there any example how to use SDO_Client   ?

CANopenNode, 2021-08-10:
Finally, example is there:
https://github.com/CANopenNode/CANopenNode/blob/master/301/CO_SDOclient.h

Should work, but, didn't test it. It should be valid for v2 and v4.

CANopenNode, 2023-03-28:
See https://github.com/CANopenNode/CANopenLinux or https://github.com/CANopenNode/CANopenDemo.
SDO communication seems to be very stable now, on both sides.