CANopenNode/CANopenNode #279: about coanopenNode release

franck48, 2021-03-19:
Hello,
in the past i use for a project CanOpenNode 3.10,
and produce a project for a 16bit pic24 microcontroller.

Today i find difficult to understand the current release : 
on the repository
https://github.com/CANopenNode/CANopenNode
i see: 301  303  309  what mean theses numbers ?  ds301?
inside each folder there is no new release of the same files but different files  ?

i dicover the new canopenPIC
https://github.com/CANopenNode/CANopenPIC
is it better to use it or to use the old canopenNode 3.10 ?

inside CanOpenPic stack, the folder CanOpen is empty, i guess we have to put inside the last canopenNode release ?.

finally i'm not sure what files and folder i have to use for a pic24 project.

thank you for helping me to understand

franck48, 2021-03-19:
i would like to "copy" all the necessary files directly inside my hard disk, inside the microcontroler project.
i prefer not to use github

ChSotiriou, 2021-03-19:
If you are downloading as zip I think you would have to also click on the CANopen submodule on github and download that as zip separately. Not sure if there is an easier way.

franck48, 2021-03-19:
i think a good idea could be this :
i create a generic project with CanopenNode for PIC24H,
with only the necessary files,
i add FreeRTOS and Run canopenNOde under FreeRTos.
then, when the compiler work i put all the source code somewhere here ?

JuPrgn, 2021-03-19:
Hi
You probably used v1.30 in the past.

Everything has been restructured in folders according to the CANopen standard that's why you see 301... 301 is CANopen base.
I will also work soon on a PIC project using newOD branch (CANopenNode V4) but at the moment CANopenPIC has not been updated for this CANopenNode version that is almost finished as mentionned here : https://github.com/CANopenNode/CANopenNode/issues/162#issuecomment-801995056

I would recommend using newOD, but there may be a few bugs that will be fixed in the coming weeks / months.

franck48, 2021-03-19:
ok i understand.
here's the link to v3.10
https://sourceforge.net/projects/canopennode/files/canopennode/CANopenNode-3.10/

CANopenNode, 2021-03-19:
3.10 on sourceforge is very old. After that project moved to github and started with version 1, then 2(master), 3 is skupped and 4 is newOD.

PICs are not yet updated to the latest version.

franck48, 2021-03-20:
when you said " 2(master) " you mean CanOpenNode can be MASTER (NMT master) ?

to help me at the begining i read very carrefully the user manual and the tutorial for the old sourceforge v1.10.
inside thes documents it's written to create master message manually using C struct.

geoffrey-vl, 2021-03-20:
forget sourceforge, the project restarted its versioning when it moved to github and it began numbering all over again from version 1. Currently, the master branch contains version 2. The newOD-branch contains version 4. There is no version 3 on github. 

Regarding NMT master, yes CanOpenNode can do that.
Read the README and peak inside that doc to get to know this stack (agan).

franck48, 2021-03-22:
Thank you all, for your clever help.
better understanding now.

even if it's old i can tell CanOpenNode 3.10 work's well.

i will follow your advice and start from the last new github release;
as i'm working with 16bit PIC24H microcontroller i will merge 
CanOpenNode and CanOpenPic and run it under Last FreeRTos son creating a project under Mplab-X and last XC16 compiler.

closed 2021-05-17 by CANopenNode