CANopenNode/CANopenNode #280: about CanOpenNode dictionnary editor

franck48, 2021-03-20:
hello all,
i just dicover a C# programm named CanOpenNode dictionnary editor

not sure about the goal of it,
is it a tool to create CO_OD.C for embeded microcontroler project ?
or, perhaps it is necessary to edit manually this file to customize the node slave project ?

geoffrey-vl, 2021-03-20:
Look for https://github.com/robincornelius/libedssharp, its a gui editor that generates your OD for CanOpenNode

franck48, 2021-03-22:
perfect,
very usefull tool, thank you Geoffrey.

i remember my first work with CANopen, a very lonely moment !!!

franck48, 2021-03-27:
about CanOpenNode dictionnary editor :
this usefull tool propose to generate source files for canopennode
  CO_OD.C and the header CO_OD.H
a parameter " TOOLS / PREFERENCES " permit to choose to generate filesz for CANOPENNODE V4 or LEGACY.
here,
https://github.com/CANopenNode/CANopenNode
it's V2.
so my question are,
witch parameter i muste choose for canopenNode v2 ?  V4 or LEGACY
where can i find canOpenNode V4 ?

once more, thank you for your help

pirupic2, 2021-04-12:
Hi , I just already tested the New Object Dicctionary with the LEGACY option and it seems it will work fine for the "Old Style" version.
Great Job¡¡¡

closed 2021-05-17 by CANopenNode