CANopenNode/CANopenNode #281: Drivers

geoffrey-vl, 2021-03-20:
It's been one year from now that the development was started to shape CONode into a more grown-up stack.
With that the drivers were also ditched, because they were not really maintained so well, because when a drvivers receive an update it was hard to verify those changes, because CONode is more like a middleware library (where drivers are not), and also because they're holding back development on the CONode stack.
I think, when we look back to that decision, it was a good choice. At least the stack now stands on its own and we can focus on improving the stack, aside of its driver.

However, I also noticed some downsides to that decision. People may be having a harder time to pick up the project and get it up and running. Since drivers are no longer part of the stack, people will need to find the [DeviceSupport page](https://github.com/CANopenNode/CANopenNode/blob/master/doc/deviceSupport.md) or else port the drivers themselves. In the end the Device Support page is just a collection of references to external projects that doesn't get a lot of pull requests to add/update drivers. It's almost as if during this one year period not a lot happened at the drivers side, which would be rather odd considering the number of forks out there and all the tickets that were handled during the year. I think this can be improved.

For example I've been working with LVGL in the past. The documentation is a real example of any open-source project out there. Like CanopNode, the library contains 'just' the middleware API's that you can rely on to make slick embedded GUI's. The drivers is mostly something you must provide yourselve or can rely on from other community projects. And the application is most certainly something you build yourselves, but they also have been examples.
Like LVGL, when you look at their github account, they have many extra repositories that demonstrate the library for all kinds of example applications and hardware boards. Some examples:
- lv_port_esp
- lv_port_stm32f746_disco
- lv_port__linux_frame_buffer
- ...

What I like about this approach is that for each board you can have your own documentation (README), pull requests, issues, releases and maintainers.
I doesn't solve all of the problems where a given drivers will no be maintained, but at least this system has some clear benefits over the current DeviceSupport page and also creates a more common ground for developers to share their code. If there is a common ground for developers to work upon then developers are also a lot more likely to share and uprove the code.

With that, I think it would be reasonable to remove any remaining drivers and  example code from the stack. I don't see any reason that some drivers would be part of the stack, while others aren't. In the end for server builds you can still pull any additional repos. I think, as it is now, it's more confusing people than helping.  Same counts for the example code, because maybe we could have multiples examples application at some point. To illustrate it would be nice for some people to see NMT master in action on MCU's.

I hope you give it some thoughts and I'd appreciate any feedback.

CANopenNode, 2021-03-22:
I agree with you thoughts.
I can maintain drivers for Linux and for PIC, because that is what I currently use. I would like to add some other microcontrollers, like STM or something, but don't have time.

I think, you idea, to "create a more common ground for developers to share their code" is a good idea. https://github.com/CANopenNode/ could be a base. There is currently https://github.com/CANopenNode/CANopenPIC repository, which contains drivers and small examples for PIC microcontrollers, which is maintained mostly by me.

I think, we can open other repositories for other microcontrollers besides PICs.  For example: someone opens an issue and propose a new repository for his device. We can discuss a little, then I create a new repo under https://github.com/CANopenNode/ and give all permissions to the developer. (Or I can just fork existing repo and add the permissions). I'm very interested for something like this.

Currently in CANopenNode are two drivers: `example`, which is blank and `socketCAN`, which is for Linux. I will improve `example` in newOD a little. I will try to make something simple, so people will quickly get the idea, how CANopenNode is working.

`socketCAN` is for Linux and is actually quite complex. It could be separated into own repo, but on the other hand it is most up to date with CANopenNode and it has most rich functionality. It is not most ideal template for porting CANopenNode to microcontrollers, but it can be used to track changes in API. For now I don't like to separate it from the base, maybe later.

I would like to keep CANopenNode/CANopenNode as thin as reasonable. It currently has most of CiA 30x standards, (except 302, which could be added). For potential device profiles (CiA 401, etc), there could be own repos.

CANopenNode/CANopenSocket was previously used for Linux example. Now it is also a Linux example, but it is not limited to Linux. `basicDevice` is could be used as a rich CANopen device with different demonstration code and could be used with other microcontrollers (demo analog/digital I/O CANopen device). `basicDevice` could also be used for some CANopen tutorials and for testing (both is partially implemented). CANopenSocket is not required to be thin, it could have several git submodules.

CANopenNode, 2021-04-07:
I have a question: in CANopenNode/socketCAN directory is driver for Linux, but there may be a confusion with CANopenSocket repository, for which I plan not to be limited to Linux, but examples for other targets may be there, etc.

I would like to rename CANopenNode/socketCAN directory to CANopenNode/driverLinux or something, to make things more clear.

geoffrey-vl, 2021-04-07:
Not to by offending, most certainly not, but I still don't see a clear benefit why for this case the driver and example code is part of CANopenNode anyway. I understand it is easier during development somehow, to verify if adjustments to the library don't break anything. But I'd separate the linux driver, call it `CANopenDriverLinux`. And make an example program, call it `CANopen-gateway-example(-linux)` which has CANopenNode and CANopenDriverLinux as submodules. You can easily do your library changes within the CANopenNode git-submodule.

In that way, CANopenSocket may disappear. Or you rename it into `CANopen-gateway-example(-linux)`. Either way, the repo name should be mostly self-explaining.

I'm also thinking of making some sort of cheap add-on shield which allows to test basic CANOpenNode programs, for example with Cia 401 profile, on real hardware. The firmware could be another repo:  `CANopen-cia401-example(-platform-boardname)`.

CANopenNode, 2021-04-07:
Maybe you are right. Linux driver together with `cocomm` and `canopend` application may be in own repo. But I don't like to have nested git-submodules, so there I would not include CANopenNode as a git submodule.

Gateway (CiA 309) is not limited to Linux, so it can stay in CANopenNode.

`CANopenNode/example` is just a simple, all-platform compile-able example and should stay there for reference.

Repository `CANopenSocket` could be renamed to `CANopenSuite` and could include submodules: CANopenNode, CANopenLinux, CANopenPIC, CANopenSTM, testing-or-other-tools, etc. It would contain examples, documentation, tutorial, testing, libedssharp, etc. It can be larger repo, but must be maintainable.

There already is `CANopenSocket/examples/basicDevice`, which should be able to compile on all target systems. It already includes basic parts from CiA401, it can be short piece of code. Target specific example code can be in own directory. This could be a reference, what we expect from driver example, which is officially included. I will do this for PIC.

If we add multiple repos `CANopen-cia401-example(-platform-boardname)`, there could be a mess, mostly because people usually don't finish and document their job. Of course, we can make an exception for well made, usable and open example device(s). But that is not just a half working example.

geoffrey-vl, 2021-04-07:
One problem I see with _a larger suite_ is that it ends up in being a collection of stuff that is not CANopenNode, and maybe not even related to each other except for the common CANopenNode. When you look at some of the recent "Github issues" in CANopenNode: they're very platform specific. So I think people are mostly interested in what applies for them, i.e. the application and platform they're using. For example, if I need a CANopen linux gateway, why would I want the STM32 code?
I probably don't even understand the platform specific stuff.
Separating things on target-solution (= use case that you want to cover, aka Cia Profile, aka application) and target-platform will at least make the code more compact and easier to grasp than having to deal with one big repo. But maybe I'm not fully understanding your point of Suite.

At work we also put each example/demo code in a separate repository. For example when we want to perform tests with specific parts of the device, or when we need to make a Proof-Of-Concept. We indeed have the problem that at some point in the future the demo code is not updated to match the latest version of the RTOS. So if you need it to work with the latest RTOS version you as a developer will need to update that example-repo. But by separating that example code at least you're assured that the demo was working at that point in time, because it was made for that specific purpose. Furthermore you can also document it for that part in time, using a README. Furthermore we're also freed of maintaining it because it is not part of something bigger "suite" kind of thing. Maintenance is delayed until the point in time where someone _needs_ the demo to be updated. Otherwise it is not updated, and that is not a problem if no one requires it to be updated. The problem with suites is that everything becomes related and you need to test everything within the suite. 

The downside is that you, as main maintainer, will need to update more repo's. However, do you really need to update them? I think the benefit for you is that you can focus on maintaining only the stuff that _you_ use. You should rely on others to update their example code, or not update at all if it can't be tested by you or someone you trust. In the end I think you personally will only need to stick to updating the CANopenLinux (driver + app) because that is what you use to unit test, and CANopenPIC because you use that as your target device. So you can focus on only those platforms.

> If we add multiple repos CANopen-cia401-example(-platform-boardname), there could be a mess, mostly because people usually don't finish and document their job. Of course, we can make an exception for well made, usable and open example device(s). But that is not just a half working example.

By separating you may end up with a dozen of repo's (at most). However, github does have a decent search engine that you can use to find repo's within an organization. Actually, I think this list will slowly grow over years. And in github the best maintained repos are shown at the top within the organization. Actually, for starters you will have less than a handful of repo's. So I guess it will be reasonable for the next couple of years. Furthermore, I don't think we'll see mass adoption of people wanting to share their example projects. Maybe for now this topic can stay open until someone has time to actually contribute his example code under the CANopenNode project. It can be some sort of working topic that can be used to share ideas on how CANopenNode example code and drivers can find a way under the CANopenNode project instead of "somewhere on the internet".

CANopenNode, 2021-04-08:
Actually I don't like to have one big repository with everything possible inside. Basically only: (generated) documentation, example with tutorial and testing, as written above. That is currently already partially included in CANopenSocket. Environment for tutorial is currently Linux, but could be extended to other systems, which would cost additional git sub-module(driver) and sub-directory under example.
This is first option of example for specific platform. In that case platform specific driver (git-submodule) only needs to provide required driver files and only simple mainline function is necessary.

If one likes to do more than that, more improved CANopen device, example or template, he can do that in own repo with own control under the CANopenNode project.

Actually these are just two options and developer of the driver may decide, which is more suitable for him. I certainly don't like to make unnecessary limitations. And also I would very like to add room for additional project/driver/example under the CANopenNode project.

I'm just trying to do something with CANopenSocket. Maybe it is best, if I just keep it as it is now and as it was in the past - a Linux (gateway) device with some additional functionalities:
- transfer Linux driver from CANopenNode into CANopenSocket or make own repo for Linux driver only.
- Linux gateway device
- Example (CiA401) device with tutorial (Linux and maybe other platforms) 
- CANopenNode testing (real CAN or virtual CAN)

CANopenNode, 2021-04-09:
I have to clarify my thoughts a little more.

Actually, if there is a driver for another platform, it must always be in own git repository (under CANopenNode project). I can give the developer all permissions for that repo and developer has freedom to choose, how to design it. I only expect some quality and finalization.

I have a plan to split Linux driver from CANopenNode into own repo. That would be small repo, just driver, mainline, simple makefile and readme, without many examples and without git submodules.

One example (blank, compileable on any platform) should stay in CANopenNode as a basic reference.

CANopenSocket will stay as is now. It is a place for example code, some documentation, several git submodules, etc, but not drivers. I will maintain it and I will try to keep it clear, but rich.

CANopenPIC is also under my maintenance. It is only a driver for three PIC families. I will add CANopenPIC as a git submodule under CANopenSocket and make a demo for PIC there, with the same application code as for Linux. But we can add demos for other platforms too.

Geoffrey, what do you think? Are my thoughts about the driver, what you expect?
CANopenSocket is actually a different topic, it is more a demo repository, I think.

geoffrey-vl, 2021-04-09:
> Actually, if there is a driver for another platform, it must always be in own git repository (under CANopenNode project). I can give the developer all permissions for that repo and developer has freedom to choose, how to design it. I only expect some quality and finalization. I have a plan to split Linux driver from CANopenNode into own repo. That would be small repo, just driver, mainline, simple makefile and readme, without many examples and without git submodules.... Geoffrey, what do you think? Are my thoughts about the driver, what you expect?

Yes, great!

> CANopenSocket will stay as is now. It is a place for example code, some documentation, several git submodules, etc, but not drivers. ...  I will add CANopenPIC as a git submodule under CANopenSocket and make a demo for PIC there

I agree on the first part, but I would not make CANopenPIC part of CANopenSocket.

> CANopenSocket is actually a different topic, it is more a demo repository, I think.

I agree, but would use it for rich OS's such as linux, not PIC.

> One example (blank, compileable on any platform) should stay in CANopenNode as a basic reference.

I see CANopenNode more as an middleware library, so I think it should not contain examples since the example code is machine agnostic anyway. It may be a bad example for "my platform". That's where the example repo's come in, they provide demo code for each "well supported" target machine. Maybe it could contain the unit testing, but as I recon that's also pretty much tied to the platform executing the code so maybe the unit testing is OK to be sitting in CANopenSocket.

CANopenNode, 2023-03-28:
CANopenNode now includes only a blank example. CANopenLinux is a separate repository with most functionalities. Near is also CANopenPIC and CANopenSTM32. CANopenDemo contains all this as git submodules, some tutorial and testing. There is also CANopenEditor, an updated fork from libedssharp.

closed 2023-03-28 by CANopenNode