CANopenNode/CANopenNode #282: OpenEDSEditor not generating OD_CNT_*** defines

ChSotiriou, 2021-03-22:
After trying generating and running custom OD.h/c files with the latest auto-build versions of OpenEDSEditor. Which version was used to successfully generate the example OD files in the example directory?

JuPrgn, 2021-03-22:
There are 2 exporter on OpenEDSEditor you can select them on Tools > Preferences. 

V4 for newOD branch
Legacy for v1.3 / V2

ChSotiriou, 2021-03-22:
I'm trying to generate a very simple NMT Master through the example (I'm not very familiar with CANopen) and the count defines are empty. There are no errors in the generation of the files.
[image]

JuPrgn, 2021-03-22:
There are probably required object missing in your eds. Try to insert a profile DS301 and export it.

ChSotiriou, 2021-03-22:
Great thank you.

CANopenNode, 2021-03-22:
There is some information in the readme,md

> If new project was started, then DS301_profile.xpd may be inserted. If existing (old) project is edited, then existing Communication Specific Parameters may be deleted and then new DS301_profile.xpd may be inserted. Alternative is editing existing communication parameters with observation to Object Dictionary Requirements By CANopenNode in objectDictionary.md.

For counters to work you have to use "Count label". See `CO_countLabel` in objectDictionary.md.

JuPrgn, 2021-03-23:
Using `DS301_profile.xpd` instead of the `.xml` when importing is working fine.

ChSotiriou, 2021-03-23:
Thank you for the info

closed 2021-03-23 by ChSotiriou