CANopenNode/CANopenNode #284: About Master Mode

muhammadt1991, 2021-03-23:
I see no example for Master mode. 
I managed to use this mode (NMT & SDOclient) to access OD of slave nodes. 
But the problem is that I do not see how to receive the logs (PDOs) from slave nodes.
Master Node does not support TPDOs nor RPDOs on its own, is that correct? 
How do I receive all PDOs from other slave nodes ?
Is there a master demo ?

geoffrey-vl, 2021-03-23:
To consume PDO's you need to map them to your device. See #270

muhammadt1991, 2021-03-24:
@geoffrey-vl 
Thanks for ur response. 
But if what you say is the only solution , it means the library is limited to small number of nodes. 
Creating OD for say 20 or 30 slave nodes (which can use up to 4 PDOs) would be a nightmare as it would result in 80 or 120 entries.
There has to be a way to report PDOs received as is.
Are you sure there is no other solution ??
for me the other alternative is to check the RX ID and forward it to application without passing through the CANopen stack.

martinwag, 2021-03-24:
> There has to be a way to report PDOs received as is.

As far as I know, no, there isn't. You can have a callback, but that also requires the PDO to be configured.

If you really have that many nodes you should pack your mappings into an config file, and in source code parse and forward that stuff to the mapping functions. You also need to somehow configure your own code on what to do with wich PDO also.

muhammadt1991, 2021-03-24:
@martinwag 
This is a huge limitation in the stack. 
The problem is that even if OD is statically created, the problem is that PDOs are variable-len, which would impose dynamic reconfiguration.
I believe I rather use the hook to forward PDOs to application directly.

martinwag, 2021-03-24:
I don't think variable lenght PDOs is within CiA301 specification, so you have to set registers on both sides if you want to change length.

Dynamic lenght leads me to the assumption you want to create some kind of serial gateway?
- have a look at this: https://www.ixxat.com/products/products-industrial/gateways-bridges/can-gw100-rs232 . This wraps serial data in a protocol on top of PDO. One byte for control, 7 bytes for data. PDO length always 8 byte
- You can also do low-level CAN communication using one of the non-used COB IDs. The functions for doing this are inside the CO_driver.h and CO_driver_target.h files. Just take a look at the stack files on how to set-up COB ID for Tx / Callback Rx.

martinwag, 2021-03-25:
PDO Length is defined in OD 0x1600+x / 0x1a00+x mappping parameters for each PDO. To change PDO length, you have to modify this mapping before. This is what the CiA specification says. If your slave device doesn't respect those, your'e on your own. You can
- either modify the stack source code
- or use the driver functions directly as mentioned before. This gives you full access to CAN bus with all functions (and all responsibility).

muhammadt1991, 2021-03-25:
Yes @martinwag  , I believe I would go with the second option. 
Nodes respect the CiA specs and they do not change their PDO len in run time. 
But it is the master that has to deal (support) different types of Nodes, i,e, static configurations (OD) is not even close to practical solution.

CANopenNode, 2021-04-07:
I think, it is not a big deal to provide 120 entries for PDO. You can make one OD object, type ARRAY of uint64_t and map all PDO to it. It is like a database. You can map up to 1 to 8 bytes to 8-byte variable. PDO without mapping is not PDO, it is custom object. With the latest version of CANopenNode (newOD) you have even more flexibility with object dictionary.

muhammadt1991, 2021-04-07:
@CANopenNode 
It is actually a big deal in my scenario. 
Nodes report different entities , so PDO can be from 1 to 8 indeed according to the entity measured (i,e temperature , humidity ... etc). 
The configuration is not Fixed, so assume u have a GW that can receive different PDOs from target nodes.
In this case , GW should have no static configurations ! , because it does not know the exact number of nodes , nor the configuration , so for example u may have configuration of 6 nodes (3 temperature & 3 humidity) or (4 temperature & 2 humidity), in this case , the GW is required to identify the node type (through accessing OD) but it should not dynamically modify the section of RPDOs, especially if it just collects data and decision making is in another layer. 
So the best in this case is to give the master mode the ability to report all received PDOs. 
It could be called GW mode or sth like that, it 's implementation unitlity and has nothing to do with CANopen Standards

CANopenNode, 2021-04-07:
Then you probably you don't need to think about your data as PDOs, they are just raw CAN messages, which PDOs basically are.
CANopenNode is just an implementation of the CANopen standard. And both are flexible enough to enable custom solutions inside.

muhammadt1991, 2021-04-07:
@CANopenNode  
That's what I did , you are absolutely right. 
I just forwarded the PDOs from driver layer to application through a callback in the CAN_module instance. 
We can mark this as closed now. 
You have any idea about the new version of this implementation ?

geoffrey-vl, 2021-04-07:
In my case, I used the LSS ID to detect the "sensor type", and implemented a custom mapping function for each sensor type that I want to support. I do this while the master is PREOP. When everything is mapped I manually set the master to OPERATIONAL.

muhammadt1991, 2021-04-07:
> In my case, I used the LSS ID to detect the "sensor type", and implemented a custom mapping function for each sensor type that I want to support. I do this while the master is PREOP. When everything is mapped I manually set the master to OPERATIONAL.

But you still have not implemented the static OD mapping , right?

geoffrey-vl, 2021-04-07:
> But you still have not implemented the static OD mapping , right?

The master hasn't mapped any PDO statically, the magic happens at runtime.

muhammadt1991, 2021-04-07:
> The master hasn't mapped any PDO statically, the magic happens at runtime.

Exactly , that's what Im doing

geoffrey-vl, 2021-04-07:
But on the 'slave' side, the mapping is fixed (currently).

muhammadt1991, 2021-04-07:
> But on the 'slave' side, the mapping is fixed (currently).

That's the case here too. 
I was not saying the target node ever changes anything !
Each node has its own pre-defined OD and PDOs.
The only things thats configurable is the Node ID (through third party utility) 
That's a CANopen Standard and should not change. 
But I'm saying I have multiple nodes (each is fixed and known) but the master (GW) does not know for example which node is reporting what at static build time. So it has to discover that on the run.

CANopenNode, 2021-04-07:
I don't understand this "static" question. OD is static, defined in compile-time. But PDO is dynamically configurable in run time, also mapping.

muhammadt1991, 2021-04-07:
@CANopenNode 
That's in case you have provided the room for the 127 entries from the beginning, and that's not what I wanna do.

closed 2021-05-17 by CANopenNode