CANopenNode/CANopenNode #287: compiler error

franck48, 2021-04-02:
hello all, i need help to understand where is my mistake :

i have created a project under MPLAB-X with the compiler XC16 for PIC24H microcontroller.

The project use CANopenPIC, inside it there is an empty folder, i add in CanOpenNode.
but during compile process i have 2 error,
the first is solved but not the 2nd

1-
the compiler said
"for loop initial declaration are only allowed in c99"
the solution was for me to declare variable outside the for loop

for(uint8_t i=0; i<HBcons->numberOfMonitoredNodes; i++)   <-- don't compile on XC16

 uint8_t i;
 for(/*uint8_t*/ i=0; i<HBcons->numberOfMonitoredNodes; i++)  <-- ok

2- (here is my problem)
compiler said this

n file included from ../CANOPEN/CANopenNode/CANopen.c:28:0:
../CANOPEN/CANopenNode/CANopen.h:272:5: error: expected specifier-qualifier-list before 'CO_TIME_t'
../CANOPEN/CANopenNode/CANopen.c: In function 'CO_new':
../CANOPEN/CANopenNode/CANopen.c:188:7: error: 'CO_t' has no member named 'TIME'
../CANOPEN/CANopenNode/CANopen.c:188:17: error: 'CO_TIME_t' undeclared (first use in this function)
../CANOPEN/CANopenNode/CANopen.c:188:17: note: each undeclared identifier is reported only once for each function it appears in
../CANOPEN/CANopenNode/CANopen.c:188:28: error: expected expression before ')' token
../CANOPEN/CANopenNode/CANopen.c:189:11: error: 'CO_t' has no member named 'TIME'
../CANOPEN/CANopenNode/CANopen.c:211:11: error: 'CO_t' has no member named 'RPDO'
../CANOPEN/CANopenNode/CANopen.c:212:15: error: 'CO_t' has no member named 'RPDO'
../CANOPEN/CANopenNode/CANopen.c:218:11: error: 'CO_t' has no member named 'TPDO'
../CANOPEN/CANopenNode/CANopen.c:219:15: error: 'CO_t' has no member named 'TPDO'
../CANOPEN/CANopenNode/CANopen.c:224:7: error: 'CO_t' has no member named 'HBcons'
../CANOPEN/CANopenNode/CANopen.c:225:11: error: 'CO_t' has no member named 'HBcons'
../CANOPEN/CANopenNode/CANopen.c:235:11: error: 'CO_t' has no member named 'SDOclient'
../CANOPEN/CANopenNode/CANopen.c:236:14: error: 'CO_SDOclient_t' undeclared (first use in this function)
../CANOPEN/CANopenNode/CANopen.c:236:30: error: expected expression before ')' token
../CANOPEN/CANopenNode/CANopen.c:237:15: error: 'CO_t' has no member named 'SDOclient'
../CANOPEN/CANopenNode/CANopen.c:244:7: error: 'CO_t' has no member named 'LEDs'
../CANOPEN/CANopenNode/CANopen.c:245:11: error: 'CO_t' has no member named 'LEDs'
../CANOPEN/CANopenNode/CANopen.c:251:7: error: 'CO_t' has no member named 'LSSslave'
../CANOPEN/CANopenNode/CANopen.c:252:11: error: 'CO_t' has no member named 'LSSslave'
../CANOPEN/CANopenNode/CANopen.c: In function 'CO_delete':
../CANOPEN/CANopenNode/CANopen.c:334:12: error: 'CO_t' has no member named 'LSSslave'
../CANOPEN/CANopenNode/CANopen.c:340:16: error: 'CO_t' has no member named 'SDOclient'
../CANOPEN/CANopenNode/CANopen.c:346:12: error: 'CO_t' has no member named 'HBcons'
../CANOPEN/CANopenNode/CANopen.c:363:16: error: 'CO_t' has no member named 'TPDO'
../CANOPEN/CANopenNode/CANopen.c:368:16: error: 'CO_t' has no member named 'RPDO'
../CANOPEN/CANopenNode/CANopen.c:373:12: error: 'CO_t' has no member named 'TIME'
../CANOPEN/CANopenNode/CANopen.c: In function 'CO_LSSinit':
../CANOPEN/CANopenNode/CANopen.c:611:30: error: 'CO_t' has no member named 'LSSslave'
../CANOPEN/CANopenNode/CANopen.c: In function 'CO_CANopenInit':
../CANOPEN/CANopenNode/CANopen.c:657:59: error: 'CO_SDOclientPar_t' undeclared (first use in this function)
../CANOPEN/CANopenNode/CANopen.c:664:26: error: 'CO_t' has no member named 'LEDs'
../CANOPEN/CANopenNode/CANopen.c:763:5: warning: implicit declaration of function 'CO_TIME_init'
../CANOPEN/CANopenNode/CANopen.c:763:26: error: 'CO_t' has no member named 'TIME'
../CANOPEN/CANopenNode/CANopen.c:830:30: error: 'CO_t' has no member named 'RPDO'
../CANOPEN/CANopenNode/CANopen.c:852:30: error: 'CO_t' has no member named 'TPDO'
../CANOPEN/CANopenNode/CANopen.c:873:32: error: 'CO_t' has no member named 'HBcons'
../CANOPEN/CANopenNode/CANopen.c:887:9: warning: implicit declaration of function 'CO_SDOclient_init'
../CANOPEN/CANopenNode/CANopen.c:887:35: error: 'CO_t' has no member named 'SDOclient'
../CANOPEN/CANopenNode/CANopen.c:889:53: error: expected expression before ')' token

it seems to come from this,  CO_CONFIG_TIME not defined

/* default configuration, see CO_config.h */
#ifndef CO_CONFIG_TIME
#define CO_CONFIG_TIME (0)
#endif
#if ((CO_CONFIG_TIME) & CO_CONFIG_TIME_ENABLE) || defined CO_DOXYGEN

in file CO_config.h i find this

#ifdef CO_DOXYGEN
#define CO_CONFIG_TIME (CO_CONFIG_TIME_ENABLE | CO_CONFIG_GLOBAL_FLAG_CALLBACK_PRE | CO_CONFIG_GLOBAL_FLAG_OD_DYNAMIC)
#endif
#define CO_CONFIG_TIME_ENABLE 0x01
#define CO_CONFIG_TIME_PRODUCER 0x02

so probably i have to activate/define  CO_DOXYGEN  but how to do this and what is it ?

thank you to help me to understand

JuPrgn, 2021-04-02:
1/ The working method to enable this is to add -std=gnu99 in Properties > XC16 (Global Options) > Additional options.
Please note that this is not documented in the XC16 User's guide (it does not appear to be an official option) and may change other things .

2/ You should clone CANopenPIC with recursive option in order to get the corresponding CANopenNode that will compile fine. This is documented on CANopenPIC readme https://github.com/CANopenNode/CANopenPIC#readme.

JuPrgn, 2021-04-02:
This PR https://github.com/CANopenNode/CANopenPIC/pull/7 that was not merged yet fix missing declarations.

franck48, 2021-04-02:
excuse me but i'm not sure to understand what files i have to replace ?

i worked like this :

1-
i downloaded the .zip file of CANOPENPIC

franck48, 2021-04-02:
i tried with this :
https://github.com/JuPrgn/CANopenPIC

but same problem ?

franck48, 2021-04-02:
tried with theses release of CanOpenPic and  CanOpenNode
some warning dissepear, seems better 
https://github.com/JuPrgn/CANopenPIC
https://github.com/CANopenNode/CANopenNode/tree/363d37b8b7b3b7fe1461562bf0b77c8071939265

i have this message from the compiler :

In file included from ../CANopenNode/CANopen.c:28:0:
../CANopenNode/CANopen.h:272:5: error: expected specifier-qualifier-list before 'CO_TIME_t'
../CANopenNode/CANopen.c: In function 'CO_new':
../CANopenNode/CANopen.c:188:7: error: 'CO_t' has no member named 'TIME'
../CANopenNode/CANopen.c:188:17: error: 'CO_TIME_t' undeclared (first use in this function)
../CANopenNode/CANopen.c:188:17: note: each undeclared identifier is reported only once for each function it appears in
../CANopenNode/CANopen.c:188:28: error: expected expression before ')' token
../CANopenNode/CANopen.c:189:11: error: 'CO_t' has no member named 'TIME'
../CANopenNode/CANopen.c:211:11: error: 'CO_t' has no member named 'RPDO'
../CANopenNode/CANopen.c:212:15: error: 'CO_t' has no member named 'RPDO'
../CANopenNode/CANopen.c:218:11: error: 'CO_t' has no member named 'TPDO'

JuPrgn, 2021-04-02:
When you download CANopenPIC you should download the corresponding submodule (CANopenNode) commit. On CANopenPIC current master, this is `CANopenNode @ aed6a47` https://github.com/CANopenNode/CANopenNode/tree/aed6a472d4c63e4b14df55ab89ff514a77f9a72f.
I would recommend using newOD branch (V4) for new designs but there may be some bugs. I recently posted a PR for newOD on CANopenPIC and this compile and works fine for me.
https://github.com/JuPrgn/CANopenPIC/tree/newOD-exampl-dsPIC33
https://github.com/CANopenNode/CANopenNode/tree/ba0b0bcdb5806241bf145adb251ca9d405b07938

franck48, 2021-04-02:
thank you,
i just check except the declaration of  "nodeId", when i change the variable name by an integer, here 10 it's ok

 err = CO_CANopenInitPDO(CO,             /* CANopen object */
                                CO->em,         /* emergency object */
                                OD,             /* Object dictionary */
                                10,//nodeId,
                                &errInfo);

i would like to ask you another question :
with this i can use canopenDictionnary editor  to generate the files OD.c and OD.h ?
and
theses release of CanOpenPic and CanopenNode are a good choice for a new project ?
(in the past i worked on CanOpenNode v3.10 from sourceforge)  and modify CO_OD.h and CO_OD.c manually.

once more thank you for helping me

JuPrgn, 2021-04-02:
Great ! 

Be sure to be on newOD branch when you are looking for documentation. Everything is documented here https://github.com/CANopenNode/CANopenNode/tree/newOD and https://github.com/CANopenNode/CANopenNode/tree/newOD/doc. 

To generate OD I use https://github.com/robincornelius/libedssharp latest build available here https://github.com/robincornelius/libedssharp/raw/gh-pages/build/OpenEDSEditor-latest.zip.

Configure OpenEDSEditor > Preferences > CANOPENNODE_V4 when using CANopenNode newOD. 
Start from a new clean project and Insert Profile `DS301_profile.xpd` note `xpd` extension and not `xml`.

closed 2023-03-28 by CANopenNode