CANopenNode/CANopenNode #289: question about CO_OD.h  and CO_OD.c  files

franck48, 2021-04-06:
hello all,
i would like to have information and help.
i work with ds33EP microcontroller , MPLAB-X and XC16.

the file  CO_OD.h  is produced by the Tool OBJECT DICTIONNARY EDITOR.
when i compile i see some compiler error
**
In file included from ../CANOPEN/CANopenNode/example/CO_OD.c:16:0:**
../CANOPEN/CANopenNode/example/CO_OD.h:35:25: error: expected '=', ',', ';', 'asm' or '__attribute__' before 'VISIBLE_STRING'
../CANOPEN/CANopenNode/example/CO_OD.h:36:25: error: expected '=', ',', ';', 'asm' or '__attribute__' before 'OCTET_STRING'
../CANOPEN/CANopenNode/example/CO_OD.h:42:25: error: expected '=', ',', ';', 'asm' or '__attribute__' before 'DOMAIN'
**
i solved it by adding this inside the file, 

**
typedef char char_t;
typedef unsigned char oChar_t;
typedef unsigned char domain_t;
**
a warning in the file indicate we do not have to modify it manually,
so perhaps my solution is not correct ?

another problem,
the compiler write this error
**
../CANOPEN/CANopenNode/example/CO_OD.c:17:21: fatal error: CO_SDO.h: No such file or directory
**
inside CO_OD.c file i can read this
**
// For CANopenNode V2 users, C macro `CO_VERSION_MAJOR=2` has to be added to project options
#ifndef CO_VERSION_MAJOR
 #include "CO_driver.h"
 #include "CO_OD.h"
 #include "CO_SDO.h"          <-------------------------------------- appear with red underline
#elif CO_VERSION_MAJOR < 4
 #include "301/CO_driver.h"
 #include "CO_OD.h"
 #include "301/CO_SDOserver.h"
#else
 #error This Object dictionary is not compatible with CANopenNode v4.0 and up!
#endif
**

inside CANopenNODE stack i don't have this file, what can i do please ?

JuPrgn, 2021-04-06:
If you are using CANopenNode newOD V4 as discussed on your previous issue your exporting with the wrong profile this is explained here :  https://github.com/CANopenNode/CANopenNode/issues/287#issuecomment-812473601

franck48, 2021-04-06:
i try and follow your advice. 
but at the end i discover the RAM memory necessary for the new OD arrangement is too heavy.
so i decide not to continue.
So i tried with the DEFAULT canopennode stack available on github,
i discover the CO_OD.c file is the same as my older project with the old v3.10 cnopennnode.
in this arrangement the file is readable by human !!, 
i prefer this for my project.
so i created a new project on MPLAB-X with CANopenPIC and this release of canopenNode.

after a lot of reading it seems the new OD is convenient for LINUX and bigger system than my small microcontroller.

So, as you indicate me, ( to forgot v3.10)
i would like to a new project with CanOpenNode Legacy OD + canopenpic.
but i have the problems described.
When i 'put' canopennode (master) inside CanOpenPIC i have some problem.

the first, as i write i have the solution to add 

typedef char char_t;
typedef unsigned char oChar_t;
typedef unsigned char domain_t;

but for the missing file, i don't know what to do 

i know too i must re-create the main file from the template found in the example folder of canopennode instead of the one inside canopenpic

JuPrgn, 2021-04-06:
On my dsPIC33 I had to configure use large code model on XC16 options. Which version are you using ? I am using the dsPIC33EP256.
If you prefer to use V2 I have working branches master on my forks too if this can help.

franck48, 2021-04-07:
i'am aware about the parametre LARGE MEMOMRY MODEL,
let me explain more :
the fork you give me last time to help me compile but only if the microcontroler owm more than 8k of RAM so 16K fro microchip.
after take a look i undestood files CO_OD became OD with new machanic.
so i decide to generate mine with the help of the C# tool but i don't find on it how to access to 0c2000 for example
or to 1f80h even after reading the doc.
sometimes when i change from a tab to another there is a freeze and my screen did not refresh.
so i would like to use the V2 object dictionnary files : CO_OD.c and CO_OD.h because on my last project i ever use it and it works well.
and ... i know how to manage manually the file i have some hand-made doc to do it.
so if you could help me wou're welcome.
i use :
MPLAB-X    v5.15
XC16           v1.61
uc               dspic33ep128mc502

franck48, 2021-04-07:
i try to do it by myself, to understand.

1- i download CANopenPIC-master
2- inside the sub-directory  canopennode i put CANopenNode-master
3-i remove the unnecessary files from other microcontroler
4-i add the option  -std=gnu99
5-inside CO_OD.c  i do this 
          #ifndef CO_VERSION_MAJOR
          #include "CO_driver.h"
          #include "CO_OD.h"
                  //#include "CO_SDO.h"                   <-------  comment this
                     #include "301/CO_SDOserver.h"  <------- add this
then i have some compiler error about undefined macro,
the probleme come from  CO_driver.h  and CO_driver_target.h.
some declaration must bu copied from the first to the second file

i continue, step by step

CANopenNode, 2021-04-07:
I didn't update CANopenPIC for a long time. It is the next thing in my CANopen schedule, together with some examples.

franck48, 2021-04-07:
i finally succeed,

i take my old explorer16 board with the dsp pod inserted,
i built then a little electronic can stage on a prototype board (specific for explorer16, buy on microchip).
i check communication with my PEAK probe.
all is ok.

if you want, i propose to send you the entire project (your project !) updated.
i do what i explain before, step by step. and a little more.

The MPLAB-X indicate on the dashboard : RAM consumed = 6k   <--- perfect for my project.

now, i will add FreeRTOS and run CanOpenNode under tasks.
without modifying CanOpenNode to have a good and clean starting MPlab project.

let me know if i can 'push' the project, i'm sure you don't need me to do it, but it could save time.

closed 2021-05-17 by CANopenNode