CANopenNode/CANopenNode #291: demo on pic24

franck48, 2021-04-09:
in a previous post a indicate to have a functionnal demo working
on explorer16 with the dsp33fj256gp710,
with the explorer16 an other PIM is delivered, so i put it on the board
the PIC24HJ256GP610A.
then i change a little the configuration bits and the processor on mplabx.
but it doesn't work,

i check with my oscilloscope the it-timer (1ms) it's ok.
i keep the same frequency as the dspic : 24mhz as FCY.

did someone have an idea ?.

i see the two can led, status and error blinking at a strange speed.

franck48, 2021-04-10:
this is the first time i see this !!

after hours of research on migration document from microchip
http://ww1.microchip.com/downloads/en/devicedoc/70637c.pdf

i finally discover an errata on the datasheet,
i use a 'buggy' pic24 .

i just replace the one i have buy a brand new bought on RS, all is ok,
CanOpenNode V2 works well on dspic33F and pic24H.

CANopenNode, 2021-04-12:
Great, thank you for the information.

closed 2021-05-17 by CANopenNode