CANopenNode/CANopenNode #292: How to verify your implementation?

Cushychicken, 2021-04-17:
This is more a question about test infrastructure than it is about CANopenNode: How do you test your implementation of CANopenNode in hardware? 

I'm working on implementing CANopenNode using an STM32F446 microcontroller, and I have been able to compile the package, expand the example driver to work with the STM32 HAL, and produce a heartbeat on a 1 second interval. All good there! I'm confident that the driver and the hardware interface are working A-OK. (Well, at least on the transmit side.) 

The thing I'm seeking to answer now is whether everything else is implemented correctly - SDO data, PDO consumer, etc. 

For example: I did a bit of extra work to embed `git describe` information into my firmware, and to tie that info to the appropriate CANopen field for manufacturer software version (if I remember right that's object 0x100A in the OD). What can I use to make a request

I'm guessing I'd need some sort of auxiliary CAN interface to my computer, and a host program to drive that CAN interface and implement the CANopen protocol. 

Some basic research shows that there are a few USB CAN interfaces out there, as well as Raspberry Pi hats that speak CAN as well. I've also found the `canopen` Python package here on GitHub that would allow for a fast implementation of CANopen for test purposes. 

Any other recommendations for how to go about testing my implementation of CANopenNode? I'm relatively new to using this protocol and I'm not sure the fastest way to test it.

geoffrey-vl, 2021-04-18:
There is a CANopenNode implementation for Linux available that you can easily build on your platform of choice (Raspberry Pi, pc, etc). As CAN interface you can go for a RPI "HAT" (which mostly features some MCP CAN chip), or a USB CAN device such as PEAK. Look [here](https://github.com/CANopenNode/CANopenSocket).
You can use that to setup a CANopen gateway on linux, and then use a client application to interface with you CANopen devices through the gateway.

Cushychicken, 2021-04-18:
Appreciate the insight, @geoffrey-vl - I usually develop on Windows, so I hadn't tried that full getting started guide on `CANopenNode` yet. This morning I spun up a Linux VM, ran thru the tutorial, and got a working virtual CAN network. It seems fairly straightforward to go from this, to a hardware interface. 

I will see about getting my hands on a hardware interface like the ones you suggest, try expanding the tutorial with hardware in the loop, and see that I'm getting expected behavior with SDO requests and the like.

Path forward seems clear - I'll leave this open for another day or two in case anyone else has a suggestion, but this is resolved in my mind. Thank you for your support!

JuPrgn, 2021-04-18:
CiA provide a test tool to check CANopen conformity based on your EDS file (SDO PDO... and checks your entries) but I am not sure if this is available to members only.

CANopenNode, 2021-04-19:
I would suggest the same as geoffrey-vl and JuPrgn. Currently I'm expanding CANopenSocket (Linux based demo, tutorial, examples, testing tools, etc.) On the other side I'm updating the CANopenPIC project, which will also have demo (Max32 board), similar as in CANopenSocket. I will add a notice about PIC32 in its tutorial too.

Currently there are only PIC microcontrollers inside CANopenNode project. But if there will be similar repositories for another microcontrollers, we have great interest to add them aside.

Cushychicken, 2021-04-25:
Closing after receiving some suggestions on how to go forward. Thanks folks!

closed 2021-04-25 by Cushychicken