CANopenNode/CANopenNode #294: CanOpenNode running on ds33E or pic24EP series

franck48, 2021-04-20:
canopennode run nice on pic24H and ds33C,
i would like to 'port' to the new familly " E ",
after reading an application note from Microchip explaining the difference between theses familly,
i try to do it,
as far as i know the ECAN module stay exactly the same,
but the DMA seems to have new feature,

does anyone have ever do this port, or can confirm the DMA must me changed ?
is there other modification to do ?

The code written in CO_driver.c is difficult to read, (define that use other define ..) difficult to follow for me.

franck48, 2021-04-21:
in my prévious message i assume Canopennode drivers works only on ds33c and pic24h/j,
it's probably an error, the driver seems to be written for the old and New DMA.
inside the file CO_driver.c we can see the author try to access to the register in a microPROCESSOR way (not microCONTROLLER)
instead of declare #DEFINE to create an alias for the register he decide to attack them by a BaseAdress and all register of the DMA is then only an ofset of these base address.

#ifndef __HAS_EDS__  //--------------------------this is the case of ds33C
    #define DMA_STA      0x4
    #define DMA_STB      0x6
    #define DMA_PAD      0x8
    #define DMA_CNT      0xA
#else  //-------------------------------------------and this is for 24EP and DS33E
    **#define DMA_STAL     0x4
    #define DMA_STAH     0x6**
    #define DMA_STBL     0x8
    #define DMA_STBH     0xA
    #define DMA_PAD      0xC
    #define DMA_CNT      0xE
#endif

one thing is strange :
on the REGISTER MAP of DS33E (for me !) DMA_STAL is 0x06  and DMA_STAH is 0x04
 **#define DMA_STAL     0x6
    #define DMA_STAH     0x4**

CANopenNode, 2021-04-21:
Can you give a link to your device datasheet. I will try to check.

franck48, 2021-04-21:
yes, (i really need help or information of what i can do)
i tried with 2 microcontroler of the "E" series from MIcrochip :

dsPIC33EP256GP506      https://www.microchip.com/wwwproducts/en/dsPIC33EP256GP506
on explorer16 board

and

PIC24EP512GU810     https://www.microchip.com/wwwproducts/en/PIC24EP512GU810
on this board    https://www.mikroe.com/mikromedia-3-pic24ep

i try with the help of this document
http://ww1.microchip.com/downloads/en/devicedoc/70637c.pdf

JuPrgn, 2021-04-21:
I am using dsPIC33EP256MC506 without any problem.

franck48, 2021-04-21:
interesting,
could you indicate me if you use the actual master code of canopennode with the last canopenpic ?

JuPrgn, 2021-04-21:
I tested V1.3 V2 and V4 and as previously discussed I have a fork for V2 and V4 CANopenPIC and pushed my PR

robincornelius, 2021-04-21:
I'm using the exact PIC24EP512GU810 with the older stack v1.3 in
production, i think i've tried v2 on it but not as far as v4 on this board.

Its been a while since i dug in this driver, but i had to do very little to
get it to work. The one big thing that sticks in my mind is something to do
with

   if(CO_CANbitRateData[i].scale == 2)
        CAN_REG(CANbaseAddress, C_CTRL1) |= 0x0800;

I think the errata for this PIC has the scale flag backwards so you end up
with wrong bit rate compared to documentation.

I may have had to add something for the CO_FSYS == 48000 as well as i PLL
mine up to the full 48Mhz

franck48, 2021-04-23:
Hello all,
first thing thank you all for your answer, advices, and help.
hers's the result of investigation, it could be helpfull for all 

my goal was to run CanOpenNode on the "new" E series microcontrollers form Microchip,
because they are more powerfull, new, and cheaper !.

as explained previously, with the help of JUPRGN i succeed to run it
on PIC24HJ256GP610A and dsPIC33FJ256GP710
it was the two P.I.M delivered with the EXPLORER-16 development board.
during work i encouter a priblem with the pic24H,
impossible to run it correctly,
as mentionned on this forum, the probleme comes from the silicium,
an older and buggy microcontroler, after reading the errata of the datasheet i see error in DMA resgisters ans E-CAN too.
after bought another PIM on FARNELL all is alright, CanOpenNode RUN perfectly.

now about the "E" series,
we found 2 kind inside it : microcontroler (pic24E) and dsp (dsPIC33E).

i started with this

PIC24EP512GU810    https://www.microchip.com/wwwproducts/en/PIC24EP512GU810
on this board            https://www.mikroe.com/mikromedia-3-pic24ep

i don't succeed tu run canopennode on it,
perhaps Mr JANEZ or Mr JUPRGN can understand why.

then return to the explorer 16 with a pim
https://www.microchip.com/developmenttools/ProductDetails/PartNO/MA330030
and as mentionned,
it does not work,
but, somme messages indicate it works,
and i find the issue :

the first problem was the pinout of the PIM,
if you look closely you will see it is very very different from the two other PIM used previously.

the second problem was to remap correctly with the PPS cantx an can1rx pin with the PPS

third, i use MCC to generate the "starting" of this dsp because i know well PIC24 but not DSP,
the CONFIGURATION BIT generated by MCC are WRONG there is a big mistake,
whatever you choose there is an error on FNOSC, it stay in INTRC even if you try to setup PRIPLL.
(try it you will see by yourself !)
so i decided to replace them with the help of MPLAB-C
windows / target memory view / configuration bits.

then, another problem comes from FREERTOS,
i run the JANEZ and JUPRGN example IO under this OS,
and it works very well for ds33F  pic24H and PIC24F,
but not for the "E" series.
FREERTOS does not exists for "E" SERIES, so microchip decide to port it
https://github.com/MicrochipTech/freeRTOS-PIC24-dsPIC-PIC32MM
it does NOT WORK !
to be sure i create only one task with a ticks at 1ms
i give it a lot of RAM and try inside it to blink a led,
the task was never RUN even only one time !

finally this morning i decide to remove it and IT WORKS !!!!
CanOpenNode works for me with all explained above.

CANopenNode, 2021-04-24:
Sometimes it takes some more effort to pass hardware issues.
I don't use 16-bit PIC devices in my projects, I only made some tests on explorer16 board with old F series.

closed 2021-05-17 by CANopenNode