CANopenNode/CANopenNode #297: Send emergerncy without affecting CANOpen state

SirCorono, 2021-05-12:
Hi
I use CANOpenNode (v2, as slave) in a project and it works really well. Thanks for your work.

I have a question regarding sending emergency messages. What I want to do:
When there is an alarm in my machine, I'd like to send an emergency to the master. The alarm is not related to CANOpen itself, there is no need to change the state back to pre-operational, it should just be an information. For that, I use a manufacturer specific error bit:
`#define CO_EM_SUB1_ERROR_STATUS     CO_EM_MANUFACTURER_START`

And call the error report function in the code:
`CO_errorReport(CO->em, CO_EM_SUB1_ERROR_STATUS, CO_EMC_DEVICE_SPECIFIC, info_code);`

This works, the emergency message is sent. However, the emergency affects the CANOpen state. It sets the errorRegister to 128 (Bit 7, manufacturer-specific) and  therefor prevents the CANOpen going into operational because of this lines in the heartbeat file:
`            case CO_NMT_ENTER_OPERATIONAL:
                if((*NMT->emPr->errorRegister) == 0U){
                    NMT->operatingState = CO_NMT_OPERATIONAL;
                }
                break;`

Is there a way to send an emergency without affecting the state of CANOpen in any way? I found in the emergency header that there are critical and informative errors, but the manufacturer specific errors are markes as info and it still affect the state:
`#define CO_EM_MANUFACTURER_START        0x30U /**< 0x30, manufacturer, info, This can be used by macros to calculate error codes */
#define CO_EM_MANUFACTURER_END          0xFFU /**< 0xFF, manufacturer, info, This can be used by macros to check error codes */`

Thanks.
Corono

CANopenNode, 2021-05-17:
Is I remember, variable "errorBehavior" in object dictionary controls this. Otherwise you can try to comment out the line `errorRegister |= CO_ERR_REG_MANUFACTURER;` in CO_Emergency.c file.

SirCorono, 2021-05-18:
Hi
Thanks for the answer. The 1029 changes the bevaviour when CANOpen is already in operational, but doens't affect the transission between pre-operational and operational:
`case CO_NMT_ENTER_OPERATIONAL: if((*NMT->emPr->errorRegister) == 0U){ NMT->operatingState = CO_NMT_OPERATIONAL; } break;`
As soon as something was written to the errorRegister, the transission doesn't work anymore. 

According to this documentation:
https://canopennode.github.io/CANopenSocket/group__CO__Emergency.html#gga587034df9d350c8e121c253f1d4eeaccaf850a661aadde65b10b22715cf24942c

CO_EM_MANUFACTURER_START | 0x30+, manufacturer, info or critical, Error status buts, free to use by manufacturer.  By default bits 0x30..0x3F are set as informational and bits  0x40..0x4F are set as critical. Manufacturer critical bits sets the  error register, as specified by CO_CONFIG_ERR_CONDITION_MANUFACTURER
-- | --

In version 2, this is not implemented. CO_CONFIG_ERR_CONDITION_MANUFACTURER is never used. In the CO_Emergency file (line 359):
`    /* Manufacturer */
    for(i=6; i<em->errorStatusBitsSize; i++) {
        if (em->errorStatusBits[i]) {
            errorRegister |= CO_ERR_REG_MANUFACTURER;
        }
    }`

For all manufacturer emergencies, the errorRegister is written and therefore they are critical. Changing the code to i=8 solves the problem for me. Maybe CO_CONFIG_ERR_CONDITION_MANUFACTURER is implemented in the newest version and the problem is solved anyway.

henrikbrixandersen, 2021-05-19:
I am facing the same problem here.

I have set the Manufacturer Specific Error Behavior (index 1029h, subindex 6) to 1h, which allows staying in Operational even if a Manufacturer Specific error is reported. It does, however, not allow transitioning from Pre-operational to Operational with a Manufacturer Specific error set.

I have been looking in CiA 301 to see what it says regarding prerequisites for transitioning from Pre-operational to Operational, but I have not been able to find any source for the `if((*NMT->emPr->errorRegister) == 0U)` check in CANopenNode.

CANopenNode, 2021-05-24:
Version v2 has very incomplete and non-standard error behavior. In v4 this has been rewritten and is configurable by `CO_NMT_control_t` parameter passed to `CO_NMT_init()` (and `CO_CANopenInit()`) functions. See https://canopennode.github.io/CANopenNode/group__CO__NMT__Heartbeat.html#gaf92cf5943801e5dda84654345cc3d67f

1029h is not used by stack, but may be used by application in connection with `CO_NMT_control_t` parameter according to the needs.

closed 2021-06-29 by CANopenNode