CANopenNode/CANopenNode #298: CANopenNode finalization

CANopenNode, 2021-05-17:
As started in #162, CANopenNode and related projects are getting final shape.

CANopenNode master branch has been switched to the latest v4.0. newOD branch has been deleted.

Under the CANopenNode ubmrella is now 7 repositories:
1. CANopenNode - base
2. CANopenDemo - tutorial and testing. It includes demoDevice, which is target independent, currently runs from Linux or PIC32.
3. CANopenNode.github.io - webpage with documentation from all repos, available https://canopennode.github.io
4. CANopenEditor - this is libedssharp fork
5. CANopenLinux - driver for Linux with "canopend" and "cocomm" applications
6. CANopenPIC - driver for Microchip PICs
7. CANopenSocket - this is now obsolete, its contents has been moved to CANopenDemo and CANopenLinux

Waiting for heroes to contribute other drivers.

Enjoy!

CANopenNode, 2021-05-17:
Some instructions about updating the branches:

To keep the v2.0 branch:

    git fetch
    git checkout v2.0-master

If you are on newOD branch and want to update to the latest v4.0, delete the old master branch first:

    git fetch
    git checkout newOD
    git branch -D master
    git checkout master

If you already did git pulling, then `git merge --abort` first.

Or just make a new clone:

    git clone https://github.com/CANopenNode/CANopenNode

geoffrey-vl, 2021-05-17:
> Waiting for heroes to contribute other drivers.
 
How to go from 
1) creating your own project with custom drivers, to 
2) projects in `deviceSupport.md`, to 
3) having it under the CANopenNode umbrella?

Maybe this info could also be added to the docs somewhere?

CANopenNode, 2021-05-18:
I updated the `deviceSupport.md`, it is mentioned there:

    7. Offer your work for inclusion under the CANopenNode project and become its developer. It will increase code quality and functionality.

Maybe now will be easier for one to prepare comparable example device. Goal may be creating a demoDevice as in CANopenDemo and running the tests there.

CANopenNode, 2023-03-28:
I'm quite happy with the transition into new version. It seems to be stable and also well accepted.