CANopenNode/CANopenNode #299: Create virtual SDO server

seifguerbouj, 2021-05-19:
I want to create a virtual SDO server using CANopenNode to communicate with my SDO client device.
Is there an example how to do that and use specific eds files related to the server that I want to create ?

m-obi, 2021-05-20:
A communication always starts by a client (master). The server (slave) just answer (serving a client request).

HayfaNc, 2021-05-23:
@m-obi  how does the server (slave) answer the client? because I think we can only read/write data in the object dictionnary using update/download functions. I'm using python to build a virtual sdo server with canopen but I didn't find any function that helps me to do so. do you please have any clue?

CANopenNode, 2021-05-24:
SDO server is already part of every device made from CANopenNode.
Here is some more information:
https://canopennode.github.io/index.html

HayfaNc, 2021-05-25:
@CANopenNode actually, I already have a client (master) but I'm working on creating a virtual canopen slave to communicate with this existing master. when running my script I keep getting this error "no sdo response recieved". As far as I know, I need to create an sdo server to send response to the master's request, is it possible to create a virtual sdo server that does the job? otherwise,how can I fix this problem?

CANopenNode, 2021-06-29:
I really don't understand the question. How CANopenNode is used in your case?

closed 2023-03-28 by CANopenNode