CANopenNode/CANopenNode #302: ballpark numbers for ROM & RAM requirements

CoderOnWinks, 2021-05-27:
Hi,

I am considering to use this stack in one of my commercial projects. Curious to see some typical figures for ROM & RAM requirements for this stack. Surely, it will vary a lot depending on enabled features and OD contents, and also on target architecture, but I think having a few ballpark numbers would be valuable to give a rough estimate of what to expect. 
It would be nice to list the smallest known, minimal device implementation requiremnts.

Thanks

geoffrey-vl, 2021-05-27:
A basic demo on mbed-os RTOS, STM32 NUCLEO L496ZG board, gcc 9 gives following build result:

```
| Module                |      .text |    .data |     .bss |
|-----------------------|------------|----------|----------|
| CANopenNode/CANopen.o |   1462(+0) |    0(+0) |   86(+0) |
| CANopenNode/stack     | 11882(+56) |    4(+4) |  408(+0) |
| CO_OD.o               |      0(+0) |  980(+0) |    0(+0) |
| [fill]                |    126(+0) |    8(+4) |   24(+0) |
| [lib]/c.a             |  28316(+0) | 2472(+0) |   89(+0) |
| [lib]/gcc.a           |   3264(+0) |    0(+0) |    0(+0) |
| [lib]/misc            |    180(+0) |    4(+0) |   28(+0) |
| main.o                |    728(+8) |    0(+0) |   94(+0) |
| mbed-os/components    |     34(+0) |    0(+0) |    0(+0) |
| mbed-os/drivers       |   1128(+0) |    0(+0) |    0(+0) |
| mbed-os/hal           |   1716(+0) |    4(+0) |   66(+0) |
| mbed-os/platform      |   3334(+0) |  260(+0) |  348(+0) |
| mbed-os/rtos          |  10140(+0) |  168(+0) | 6289(+0) |
| mbed-os/targets       |  12130(+0) |    4(+0) |  888(+0) |
| Subtotals             | 74440(+64) | 3904(+8) | 8320(+0) |
Total Static RAM memory (data + bss): 12224(+8) bytes
Total Flash memory (text + data): 78344(+72) bytes
```

Note that mbed takes a considerable amount of memory, but I'm also using C++ standard libs every now and then which also adds up a lot. Some guys over here are running on a much less powerful board.

CANopenNode, 2021-05-28:
I will try to configure v4 with basic features for one old device with 2k RAM. Not sure, but may work.

martinwag, 2021-05-31:
ARM Cortex M4F (STM32F3)
- fully featured FreeRTOS + Drivers + most features of stack, without application code ~50kB
- Bootloader without OS, stripped drivers, stripped stack ~13kB