CANopenNode/CANopenNode #308: OD Extensions on Subindex Items

StormOli, 2021-05-28:
How would I go about adding different OD Extensions for different sub index items. Take the following function for example

```
int add_od_extension(uint16_t index, uint8_t subindex, OD_extension_t *od_extension,
                        ODR_t (*read)(OD_stream_t *stream, void *buf,OD_size_t count, OD_size_t *countRead), 
                        ODR_t (*write)(OD_stream_t *stream, const void *buf, OD_size_t count, OD_size_t *countWritten))
{
    ODR_t odRet;
    OD_entry_t *entry;
    OD_IO_t odIO;
    entry = OD_find(OD, index);
    odRet = OD_getSub(entry, subindex, &odIO, false);
  
    od_extension->object = odIO.stream.object;
    od_extension->read = read;
    od_extension->write = write;        
 
    OD_extension_init(entry, od_extension);
  
    return 0;
}
```

If I pass in 8 different subindex's but the same main index, All my OD_entry_t will be identical.  Although I'll end up with 8 unique OD_Extensions, every time I run OD_extension_init it will simply overwrite the previous extension for that "entry".
Thus no mater what sub index I access, it will always attempt to run the last method passed to the main index.

CANopenNode, 2021-05-29:
You can have extension for each index, not sub-index. Use `switch (stream->subIndex) {`.

See https://github.com/CANopenNode/CANopenDemo/blob/master/demo/objectAccessOD.c

StormOli, 2021-05-29:
Thanks Janez,
    I think I got caught up in the belief that extension-> object had to be the entry stream object, but looking at your demo it looks like it's only used in the extensions custom read and write functions and is completely optional?
If so that makes things much clearer in my head and I can easily accomplish what I require.

StormOli, 2021-06-02:
Got this coded up and working perfectly. Thanks again for your help Janez

closed 2021-06-02 by StormOli