CANopenNode/CANopenNode #310: possibility of sdo upload big files over coconn domain type  > 10 MByte

apiller-jetter, 2021-06-16:
Is the new coconn in principle able to upload larger files?

CANopenNode, 2021-06-29:
Yes, I tested 10GB in virtual CAN interface in Linux, few hours, no problems. Block transfer is much more efficient, segmented transfer works too. See https://canopennode.github.io/md_test_test_report.html

apiller-jetter, 2021-06-30:
Thanks for the hint. 
I had the problem that our SDO server (receiver) needs a stream size information. 
Unfortunately cocomm does not have this option, so I have extended the "CO_gateway_ascii.c" with a new command. 
"[<net>] set sdo_wr_size <value>          # Set default sdo write size.\n" \
I don't know that the best solution is, but it works. 
Does it make sense to enter this into the master?
I could upload to a brach, what do you mean?

CANopenNode, 2021-06-30:
I don't like to add non-standard solutions.

apiller-jetter, 2021-06-30:
I understand that.
How can I then send a file with a size information transfer via cocomm/SDO. 
Are there other options?

CANopenNode, 2021-06-30:
Size information is part of the SDO message, it is optional. Actually cocomm don't know the size information of large data on the beginning of transfer (write). So maybe your solution may be simple and useful. Pleas make a pull request then.

closed 2023-03-28 by CANopenNode