CANopenNode/CANopenNode #311: Not possible to compile latest master branch

MaJerle, 2021-06-24:
For some reason, in GCC for ARM, it is not possible to compile the lib.

Errors like "CO_SDO_t" or _CO_storae_entry_t_ errors are appearing in the core lib.

Why?

CANopenNode, 2021-06-24:
Živijo,

Missing include directories?
Can you try:
```
cd CANopenNode/example
make
```
Note also, latest master branch switched to v4.0 recently, it is quite an upgrade (new object dictionary, new SDO server, etc.), see https://github.com/CANopenNode/CANopenNode/issues/298

MaJerle, 2021-06-24:
Janez, many thanks for your feedback. Seems you are Slovenian, or at least speak Slovenian :) Hvala za odgovor.

Currently I have a working solution with PIC old examples and will continue with this one.
I would appreciate to get your email for contribution and example on STM32 platform.

Let me know if you are interested, or contact me at tilen@majerle.eu

closed 2023-03-28 by CANopenNode