CANopenNode/CANopenNode #313: Best Way To Create Dynamic RPDOs

StormOli, 2021-06-25:
I have a network with 1 consumer and an unknown number of producers.
I can scan the network to get the device type and node IDs of any active nodes, at which point I also know what TPDOs they are producing.  All the nodes have a maximum of the default 4 COB +NodeID TPDOs.
I need to be able to set the matching RPDOs on the consumer but am unsure the best way to do this?

1) Should I have all possible default COB-IDs pre mapped as Communication Parameters with blank matching mapping parameters at compile time, or can I add to the Object Dictionary after the network is running and I scan the nodes?

2) After I've scanned the nodes and am ready to rewrite the mapping parameters, is this done via a function somewhere or do I just edit the OD directly?
Thanks

StormOli, 2021-06-26:
Thinking about this I think it's outside the scope of CANOpen so I'm tempted to just redirect all PDO messages to some custom logic instead of copying to the CANopenNode stack, just for the 1 "gateway/master" device.
Does this sound like the preferred option?

closed 2021-06-28 by StormOli

CANopenNode, 2021-06-29:
RPDO consumer is configured via OD parameters, I didn't provide additional interface. There are several options: 
- external application scans nodes and uses gateway to configure OD parameters of gateway or other nodes
- use SDO client to configure own or remote OD parameters
- call CO_CANopenInitPDO() after scanning remote nodes and editing own `OD_PERSIST_COMM.x16xx_RPDOMappingParameter` variables from OD.

This is custom solution, I think