CANopenNode/CANopenNode #315: Mapping Boolean PDO

StormOli, 2021-06-28:
How would I go about mapping a Boolean datatype in a PDO?
With the following code

```
/* verify access attributes, byte alignment and length */
    OD_attr_t testAttribute = isRPDO ? ODA_RPDO : ODA_TPDO;
    if ((OD_IOcopy.stream.attribute & testAttribute) == 0
        || (mappedLengthBits & 0x07) != 0
        || OD_IOcopy.stream.dataLength < mappedLength
    ) {
        
        return ODR_NO_MAP; /* Object cannot be mapped to the PDO. */
    }

```
(mappedLengthBits & 0x07) != 0 means the PDO must be a multiple of 8 (Whole Bytes).  Does this preclude the use of the Boolean data type in CANopenNode?  If so I'll simply change the types to uint8's but thought I'd ask the question as BOOLEAN is a CANopen datatype.

CANopenNode, 2021-06-29:
Yes, CANopenNode is limited to whole bytes (granularity). It would be too complex to map each bit.

apiller-jetter, 2021-06-30:
Too bad.... PDO is only available in a limited number, therefore we needed a bitwise mapping. Are you  open for a expansion. I can make a suggestion.

CANopenNode, 2021-06-30:
There is 512 RPDOs and 512 TPDOs available. Your application can combine multiple booleans to byte. I don't understand the problem.

apiller-jetter, 2021-06-30:
My problem is that we have I/O modules with a lot of digital I/O's.  If we need a byte for each digital input or output, then we also consume quite a bit of CAN BUS bandwidth.  
Combining them on the application side does not work for me, because then I will not be able to freely map the bits. 
DS 301 mapping also provides that the PDO variable sizes are specified in bits. 
For us that would be a very important feature, 
Are you interested in working with us?

CANopenNode, 2021-06-30:
I understand.
I'm afraid, bit sized mapping may complicate the things and consume much system resources. We could implement this, but I would like to be configurable via CO_CONFIG_PDO_xxx macro. 
I currently don't have much time to cooperate. I like every improvement of the stack and I can offer some conversation, etc.

StormOli, 2021-07-01:
> Yes, CANopenNode is limited to whole bytes (granularity). It would be too complex to map each bit.

Thanks Janez,
    That's fine as moving from bit to byte is not an issue for me and it all ends up as a byte on the C side anyway so I wouldn't be saving memory on the device.  
I do however understand apiller's issue with I/O and presumably that would extend to wanting more than 8 mappings per PDO as well?
Consider this issue closed for me but I'll leave it open so @apiller-jetter  can carry on the conversation with you.

closed 2023-03-28 by CANopenNode