CANopenNode/CANopenNode #321: CO_NMT_Heartbeat.c still has old style "CO_CONFIG_FLAG_TIMERNEXT"

wilkinsw, 2021-07-16:
https://github.com/CANopenNode/CANopenNode/blob/3a0039716bceb1ea377c41014d426d47f100b362/301/CO_NMT_Heartbeat.c#L311

Should be "CO_CONFIG_GLOBAL_FLAG_TIMERNEXT"

wilkinsw, 2021-07-16:
It looks like there are several other places as well:
  ```
  $ grep -re CO_CONFIG_FLAG_TIMERNEXT
  303/CO_LEDs.c:#if (CO_CONFIG_LEDS) & CO_CONFIG_FLAG_TIMERNEXT
  301/CO_SDOserver.c:#if (CO_CONFIG_SDO_SRV) & CO_CONFIG_FLAG_TIMERNEXT
  301/CO_SDOserver.c:#if (CO_CONFIG_SDO_SRV) & CO_CONFIG_FLAG_TIMERNEXT
  301/CO_SDOserver.c:#if (CO_CONFIG_SDO_SRV) & CO_CONFIG_FLAG_TIMERNEXT
  301/CO_Emergency.c:  #if (CO_CONFIG_EM) & CO_CONFIG_FLAG_TIMERNEXT
  301/CO_config.h:#define CO_CONFIG_FLAG_TIMERNEXT 0x2000
  301/CO_config.h:/** This flag may be set globally to @ref CO_CONFIG_FLAG_TIMERNEXT */
  301/CO_config.h: * - #CO_CONFIG_FLAG_TIMERNEXT - Enable calculation of timerNext_us variable
  301/CO_config.h: * - #CO_CONFIG_FLAG_TIMERNEXT - Enable calculation of timerNext_us variable
  301/CO_config.h: * - #CO_CONFIG_FLAG_TIMERNEXT - Enable calculation of timerNext_us variable
  301/CO_config.h: * - #CO_CONFIG_FLAG_TIMERNEXT - Enable calculation of timerNext_us variable
  301/CO_config.h: * - #CO_CONFIG_FLAG_TIMERNEXT - Enable calculation of timerNext_us variable
  301/CO_config.h: * - #CO_CONFIG_FLAG_TIMERNEXT - Enable calculation of timerNext_us variable
  301/CO_config.h: * - #CO_CONFIG_FLAG_TIMERNEXT - Enable calculation of timerNext_us variable
  301/CO_config.h: * - #CO_CONFIG_FLAG_TIMERNEXT - Enable calculation of timerNext_us variable
  301/CO_config.h: * - #CO_CONFIG_FLAG_TIMERNEXT - Enable calculation of timerNext_us variable
  301/CO_PDO.c: #if (CO_CONFIG_SYNC) & CO_CONFIG_FLAG_TIMERNEXT
  301/CO_PDO.c:   #if (CO_CONFIG_PDO) & CO_CONFIG_FLAG_TIMERNEXT
  301/CO_PDO.c: #if (CO_CONFIG_PDO) & CO_CONFIG_FLAG_TIMERNEXT
  301/CO_SYNC.c: #if (CO_CONFIG_SYNC) & CO_CONFIG_FLAG_TIMERNEXT
  301/CO_SYNC.c:#if (CO_CONFIG_SYNC) & CO_CONFIG_FLAG_TIMERNEXT
  301/CO_SDOclient.c:#if (CO_CONFIG_SDO_CLI) & CO_CONFIG_FLAG_TIMERNEXT
  301/CO_SDOclient.c:#if (CO_CONFIG_SDO_CLI) & CO_CONFIG_FLAG_TIMERNEXT
  301/CO_SDOclient.c:#if (CO_CONFIG_SDO_CLI) & CO_CONFIG_FLAG_TIMERNEXT
  301/CO_SDOclient.c:#if (CO_CONFIG_SDO_CLI) & CO_CONFIG_FLAG_TIMERNEXT
  301/CO_SDOclient.c:#if (CO_CONFIG_SDO_CLI) & CO_CONFIG_FLAG_TIMERNEXT
  301/CO_SDOclient.c:#if (CO_CONFIG_SDO_CLI) & CO_CONFIG_FLAG_TIMERNEXT
  301/CO_SDOclient.c:#if (CO_CONFIG_SDO_CLI) & CO_CONFIG_FLAG_TIMERNEXT
  301/CO_HBconsumer.c:#if (CO_CONFIG_HB_CONS) & CO_CONFIG_FLAG_TIMERNEXT
  301/CO_NMT_Heartbeat.c:#if (CO_CONFIG_NMT) & CO_CONFIG_FLAG_TIMERNEXT
  304/CO_SRDO.c:#if (CO_CONFIG_SRDO) & CO_CONFIG_FLAG_TIMERNEXT
```

CANopenNode, 2021-07-22:
Actually it is correct.
`CO_CONFIG_GLOBAL_FLAG_TIMERNEXT` is by default set to 0. Alternatively may be set to the value equal to `CO_CONFIG_FLAG_TIMERNEXT`. 

`CO_CONFIG_GLOBAL_FLAG_TIMERNEXT` is used in header files for default value of `CO_CONFIG_NMT`, for example.

`CO_CONFIG_FLAG_TIMERNEXT` is used in c source files.

wilkinsw, 2021-07-22:
I see, thanks for clarifying!

closed 2021-07-22 by wilkinsw