CANopenNode/CANopenNode #322: NMT state (master)

JuPrgn, 2021-07-23:
Hi,
I need confirmation on NMT transitions:
My device is configured to start automatically in Operational.
When the device is powered up or after a reset, it sends an HB 700 + NodeID with the value 0x00 (Initialization) and goes into Operational without sending another HB to inform of this change.
I guess the device should also send an HB 700 + NodeId with the value 0x05 (Operational), right?

CANopenNode, 2021-07-23:
Bootup is always sent. But HB producer is disabled by default. Use 1017 object.

JuPrgn, 2021-07-23:
Ok I was wondering if this transition was part of the Bootup or not, thanks Janez !
I switched to the latest master on CANopenNode and PIC and I erased some of my previous config changes that I probably did not put on the right place...

closed 2021-07-23 by JuPrgn