CANopenNode/CANopenNode #323: EMCY msg not sent sometimes

JuPrgn, 2021-07-27:
With latest master, using CANopen Conformance Test Tool I get errors with EMCY Tests, sometimes EMCY msg is not sent and test fails.

I run the same test sequence and the results are random : 

No EMCY : Fail : 
```
Test Info	: ### Test case : 5.4  -  Test name : Emergency 04 ###############################
Test Info	: Force transmission of emergency message
Test CAN	: CAN1  7FF Tx SD  6  01 05 04 00 00 00
Test Info	: Emergency 04 started
Test CAN	: CAN1  603 Tx SD  8  40 00 16 00 00 00 00 00
Test CAN	: CAN1  583 Rx SD  8  4F 00 16 00 08 00 00 00
Test Info	: use PDO 1

Test CAN	: CAN1  000 Tx SD  2  01 03
Test CAN	: CAN1  603 Tx SD  8  40 00 14 01 00 00 00 00
Test CAN	: CAN1  703 Rx SD  1  05
Test CAN	: CAN1  583 Rx SD  8  43 00 14 01 03 02 00 00
Test CAN	: CAN1  603 Tx SD  8  40 00 14 02 00 00 00 00
Test CAN	: CAN1  183 Rx SD  6  02 00 00 00 00 00
Test CAN	: CAN1  583 Rx SD  8  4F 00 14 02 FF 02 00 00
Test CAN	: CAN1  203 Tx SD  0 
Test CAN	: CAN1  000 Tx SD  2  80 03
                                                                        <-- Missing EMCY
Test CAN	: CAN1  703 Rx SD  1  7F
```

EMCY Ok : Pass : 
```
Test Info	: ### Test case : 5.4  -  Test name : Emergency 04 ###############################
Test Info	: Force transmission of emergency message
Test CAN	: CAN1  7FF Tx SD  6  01 05 04 00 00 00
Test Info	: Emergency 04 started
Test CAN	: CAN1  603 Tx SD  8  40 00 16 00 00 00 00 00
Test CAN	: CAN1  583 Rx SD  8  4F 00 16 00 08 00 00 00
Test Info	: use PDO 1

Test CAN	: CAN1  000 Tx SD  2  01 03
Test CAN	: CAN1  603 Tx SD  8  40 00 14 01 00 00 00 00
Test CAN	: CAN1  703 Rx SD  1  05
Test CAN	: CAN1  583 Rx SD  8  43 00 14 01 03 02 00 00
Test CAN	: CAN1  603 Tx SD  8  40 00 14 02 00 00 00 00
Test CAN	: CAN1  183 Rx SD  6  02 00 00 00 00 00
Test CAN	: CAN1  583 Rx SD  8  4F 00 14 02 FF 02 00 00
Test CAN	: CAN1  203 Tx SD  0 
Test CAN	: CAN1  000 Tx SD  2  80 03
Test CAN	: CAN1  083 Rx SD  8  10 82 00 04 00 00 00 00    <-- EMCY
Test CAN	: CAN1  703 Rx SD  1  7F
```

martinwag, 2021-07-27:
I've tried verifying with the test tool (and V1 stack) a few years ago. The quality for some of the test cases at this time was quite bad (either because no thought was given or testing would have needed physical/bit level access to CAN bus). I don't know about current state or current stack version.

regarding emergencies I've made those notes back then:
- disable heartbeat (write "0")
- test tool tries to trigger error with invalid RPDO and bus errors. RPDO errors are silently discarded by stack, bus errors are only sometimes processed correctly.

I reccomend you to try to understand how the individual test cases work, then you know if you can igore it or need investigation :-)

JuPrgn, 2021-07-27:
Hi Martin

Thank you for your answer.

With latest stack and my current configuration,  all the tests pass except 1 that may be related to a bug in this test tool when there are InfoCode on EMCY msg (I contacted CiA to clarify this) but at the moment I set this InfoCode to 0 on this specific test.

The problem I report here seems to be a bug with current EMCY implementation, this is probably time related as it is random using the same sequence and I double checked that this is not related to test tool with another CANopen sniffer on the bus.

JuPrgn, 2021-07-27:
This is probably what you observed with your note  : RPDO errors are silently discarded by stack. 
I thought it was detected and the EMCY was set without sending msg but I will check if this is not detected at all.

JuPrgn, 2021-07-27:
Test tool change NMT state to PreOp before the EMCY is sent so this code is not executed if state changed before `CO_RPDO_process()` call : 

```
    if (PDO->valid && NMTisOperational
#if (CO_CONFIG_PDO) & CO_CONFIG_PDO_SYNC_ENABLE
        && (syncWas || !RPDO->synchronous)
#endif
    ) {
        /* Verify errors in length of received RPDO CAN message */
        if (RPDO->receiveError > CO_RPDO_RX_ACK) {
            bool_t setError = RPDO->receiveError != CO_RPDO_RX_OK;
            uint16_t code = RPDO->receiveError == CO_RPDO_RX_SHORT
                          ? CO_EMC_PDO_LENGTH : CO_EMC_PDO_LENGTH_EXC;
            CO_error(PDO->em, setError, CO_EM_RPDO_WRONG_LENGTH,
                     code, PDO->dataLength);
            RPDO->receiveError = setError
                              ? CO_RPDO_RX_ACK_ERROR : CO_RPDO_RX_ACK_NO_ERROR;
        }
```

We can run this code even if we are no longer in the Operational state to notify that there has been an error before NMT change. But I don't know if I have to keep SYNC checks and if it is also required to check PDO-> valid for this or if I can safely move this code out of this if ?

CANopenNode, 2021-08-09:
Last week I was testing with CANopen Conformance Test Tool, version 3.0.3.10, latest stack, MAX32 board from CANopenDemo.
There were some small errors, which are fixed with the latest commits.

Everything was successful or skipped, except:
 - Test case : 1.0 (EDS checker 01): Check EDS file
   - Some warnings: Access type in section xxxx has no clear mapping direction.
 - Test case : 3.7 (PDO 07): Configure TPDO for sync (acyclic) mode, if supported (Did not find PDO traffic, but should)
   - This is acyclic mode, one PDO is sent, but second isn't, because there is no change-of-state. Failure is still reported.
 - Test case : 3.22 (PDO 22): Try to write a new correct mapping with the mapping procedure.
   - Failure is reported, no reason indicated, erroneous behavior wasn't recognized.
 - Test case : 4.2 (OD 02), Test case : 4.8 (OD 08): Compare the read value against the default value in the EDS if it exists.
   - There are some differences between the values in some objects.
 - Test case : 5.6 (Emergency 06): read object 1003:1 and compare it with received msg (Additional information 0x42 doesn't match)
   - Additional information are manufacturer specific anyway
 - Test case : 9.4 (State 04): Test if DUT supports NMT state stopped correctly.
   - Test sometimes fails: CCT tool detects SYNC message immediately after NMT command stop. NMT status may not be updated fast enough. However, on another CANdump terminal there is no SYNC after NMT stop command.

I think we can ignore those errors.

----
I never get error on Emergency 04 test. PDO and emergency works in different threads, so this emergency simulation may not be perfectly suitable for our case. But I think, there is no conformance violation. I think, it is not necessary to change the CO_RPDO_process() function.

JuPrgn, 2021-08-09:
Thank you for your tests ! 

I did not found conformance violations too. You did a great job ! =)

You are right test 5.6 is not an issue it is on conformance test tool side as only first bytes should be compared.

Moving previous error notification outside of "NMT state check" fix test EMCY 04 for me but this is a really specific test and time related so no problem if you don't change it.

closed 2021-08-09 by JuPrgn

JuPrgn, 2021-08-09:
I add here this note for myself to check this later, this is an answer from CiA based on the log I sent : 

The PDO error 8210h indicates device is in error state. Despite that er = 00h in the EMCY generated by your device. It is an error preventing the device from successful certification and passing the EMCY test (see EMCY 04, 05 and 06). The ER value shall in case of error contain at least Bit 0 =1 (Error state machine -> device is in error state), further bits can be set (see object 1001h). The reason for the test to verify correct ER because in EMCY 04 it is always an error. According to whatever additional error condition is available ER has value +1. You can also see it with EMCY 05 which reads 1001 back but returns 0 which is wrong given the error occured in the test EMCY 04.

Another minor issue is this value 02h appearing during reading various UNSIGNED8 objects (EMCY 05, 1600 00h and other tests). Although the further Bytes are not evaluated it should be taken care of.

CANopenNode, 2021-08-09:
> Another minor issue is this value 02h appearing during reading various UNSIGNED8 objects (EMCY 05, 1600 00h and other tests). Although the further Bytes are not evaluated it should be taken care of.

This was a bug. [Fixed memcpy](https://github.com/CANopenNode/CANopenNode/commit/b4ddf9be32450e28685b2a48119a327ce51651ef?branch=b4ddf9be32450e28685b2a48119a327ce51651ef&diff=split#diff-5af93e722e3f8a448c9f1e708bd5a8cc2657b1aaee136df37cd168d23e7b3402L1265-R1267)