CANopenNode/CANopenNode #324: Support for multiple CAN nodes on 1 physical CAN interface?

ThomasGerstenberg, 2021-08-06:
I have a use-case where I have multiple motors being driven by a single microcontroller (STM32F4) with a single CAN tranceiver. Each motor would implement CiA 402, which I think means that each motor would require its own Node ID and associated object dictionary.

Is this something that this library can support, or is there an alternative solution I'm unaware of (relatively new to CANOpen)? If it is supported, what would the solution look like? 

I've done an initial dive into the code I think it looks like one `CO_t` object per OD all sharing the same `CO_CANmodule_t` object, since it seems the Node ID is tied to the `CO_t` and not the OD. Please correct me if I'm wrong :)

ThomasGerstenberg, 2021-08-06:
Did a deeper dive into CiA 301 and 402 and how PDOs work. Looks like 301 specifies up to 8 logical devices within a single object dictionary, the main issue is that only 4 TPDO and RPDOs are given standardized COB-ID function codes. 301 specifies up to 512 PDOs, meaning the remaining 508 COB-IDs for PDOs are manufacturer-specific.

Let's say for my board I use nodeID+0...nodeID+7 for PDO mapping. Is there a way to configure those PDO communication parameters at runtime for the non-standard PDO entries?

CANopenNode, 2021-08-09:
I think, up to 8 logical devices is best solution for your case. PDOs are not the problem. There are 4 TPDO and RPDOs, which just have pre-defined COB-IDs. You can have any number of PDOs on single device, with one nodeID and you can configure theirs COB-IDs to anything. Just make a good plan for your devices, which node-IDs each will have (for example 8, 16, 24, etc) and configure their PDOs to sensible values.

ThomasGerstenberg, 2021-08-09:
Thanks for your response! Sounds like I'm thinking of the same approach.

I'm planning on allocating 4 TPDO/RPDOs per logical device so I can reuse the TPDO/RPDO function codes with incremented node ids. Then each physical device would be separated by 8 node IDs like your example .

Just to confirm, do the below steps look correct?

1. determine the node ID at runtime (jumpers/switches/etc)
2. update the PDO COB-IDs within the object dictionary for PDOs 5+
3. call `CO_CANopenInitPDO()`

CANopenNode, 2021-08-09:
I think, this can be a good approach. PDOs will be automatically configured on startup. User could re-configure them in run-time, if he like so. If you will use eeprom storage for the parameters, it may be a little more complicated.

CANopenNode, 2021-08-10:
Another, most flexible and elegant option for automatic (more than four) default PDOs when using logical devices may be extending `preDefinedCanId` argument in CANopen.c file:
```
            uint16_t preDefinedCanId = i < 4
                                     ? (CO_CAN_ID_TPDO_1 + i * 0x100) + nodeId
                                     : 0;
```
I will try to make "4" and equation configurable.

CANopenNode, 2021-08-10:
I added that, because I think, it could be a useful option for logical devices. See https://github.com/CANopenNode/CANopenNode/commit/b1b091a46330fdc82a3d5106a2976879b903e0ae

You have to define own `CO_RPDO_DEFAULT_CANID_COUNT` and `CO_TPDO_DEFAULT_CANID_COUNT` constants somevhere from CO_driver_target.h file. In Object dictionary editor add T(R)PDOs, with default value, for example TPDO1->COB_ID=TPDO5->COB_ID=0x180. And Node-IDs will be added automatically.

Note, device still have single node-id. But PDOs occupies additional CAN addresses. So there should be a node-id gap between devices.

Please respond, if everything will work as expected.

ThomasGerstenberg, 2021-08-10:
This is perfect, thanks!
I'll be working on it this week and let you know how it goes 👍

ThomasGerstenberg, 2021-09-14:
This solution works on the FW side, however we're running into issues on the controller side (using python canopen). 

The python lib reads the COB-ID from the EDS, so RPDO's are defined as `0x200+$NODEID` and uses that $NODEID flag to add the value in. For RPDO 5 the EDS would ideally contain `0x201+$NODEID`, however the current OD.c doesn't retain knowledge of that `$NODEID` flag. `CO_RPDO_init()` sees that the node portion of the COB-ID is nonzero and leaves it as-is.

I think the ideal solution would be for OD.c to retain knowledge of that `$NODEID` flag. Could it work as a bit flag in the unused bits of the COB-ID? Likely would require an update to CANopenEditor to set the bit