CANopenNode/CANopenNode #326: Infinite loop in CO_LEDs_process

ConductedForce, 2021-08-06:
I imagine this isn't normal but I wonder if I have a config issue somewhere.

CANopenNode, 2021-08-09:
Can you explain more precisely?

ConductedForce, 2021-08-10:
Within the `CO_LEDs_process()` function, it looped for a seemingly unlimited time. Right inside the `while (LEDs->LEDtmr50ms >= 50000)` section. I didn't spend much time on it as I want to try and get the core of CANopenNode to run on my device first.

CANopenNode, 2021-08-10:
But there is also inside the loop:
`LEDs->LEDtmr50ms -= 50000;`

ConductedForce, 2021-08-10:
When I had debugged it initially, it had near max value of a 32 int. Something over 4 million. It makes me wonder if I have an initial variable incorrectly set somewhere. It runs on the ticktimer for an STM32 where CO_process() runs every 1 ms.

CANopenNode, 2021-08-10:
Maybe `CO_LEDs_init` have been missed. There is `memset(LEDs, 0, sizeof(CO_LEDs_t));`

CANopenNode, 2021-08-10:
Or there was very large initial `timeDifference_us`.

ConductedForce, 2021-08-10:
CO_LEDs_init is not present, which seems to be the issue, I will be testing it shortly.

ConductedForce, 2021-08-10:
Ah, it is called but I passed over it.

ConductedForce, 2021-08-10:
LEDtimer50ms is 0 before and after init. At some point before the first CO_LEDs_process, it becomes 4122282055. Everytime.

CANopenNode, 2021-08-10:
Pointer escaped somewhere?

ConductedForce, 2021-08-10:
I've been taking from the example and another CANopenNode port, which may be the issue. I have a 1ms time loop from a ported example and then I do what's in the while loop of main_blank.c which is 99.99% of the problem. Changing timeDifference_us to 1000 should patch it.

ConductedForce, 2021-08-10:
That didn't fix it. It still has a large value and I can't find anywhere where it would be reset.

ConductedForce, 2021-08-10:
I'm not certain what's wrong with my setup or configuration but its copied straight from the example directory, minus and extra 1ms timer loop. Have to move on from it.

closed 2021-08-10 by ConductedForce