CANopenNode/CANopenNode #330: Access to received CAN message

IhorNehrutsa, 2021-08-20:
It seems that this code 
https://github.com/CANopenNode/CANopenNode/blob/9d1e99f66e34840adab90ad0dbeb23fc19a68c87/example/CO_driver_target.h#L62-L66

may be replaced by 
```
/* Access to received CAN message */
#define CO_CANrxMsg_readIdent(msg) ((uint16_t)(((CO_CANrxMsg_t *)(msg))->ident))
#define CO_CANrxMsg_readDLC(msg)   ((uint8_t)(((CO_CANrxMsg_t *)(msg))->DLC))
#define CO_CANrxMsg_readData(msg)  ((uint8_t *)(((CO_CANrxMsg_t *)(msg))->data))
```
from CANopenPIC
https://github.com/CANopenNode/CANopenPIC/blob/f54d8f89ee28fce0e1e33b1575af171046594a65/PIC32/CO_driver_target.h#L100-L104

CANopenNode, 2021-08-20:
In CANopenNode/example/CO_driver_target.h there is no CO_CANrxMsg_t structure, as it is not required by the stack.

For more description see also https://github.com/CANopenNode/CANopenNode/blob/master/301/CO_driver.h

IhorNehrutsa, 2021-08-20:
In this case compile error in line # 493 
**argument 2 null where non-null expected [-Wnonnull]**
because **data** variable is NULL
https://github.com/CANopenNode/CANopenNode/blob/9d1e99f66e34840adab90ad0dbeb23fc19a68c87/301/CO_PDO.c#L465-L493

CANopenNode, 2021-08-20:
You have to write correct macros for your microcontroller.

IhorNehrutsa, 2021-08-20:
>> In CANopenNode/example/CO_driver_target.h there is no CO_CANrxMsg_t structure, as it is not required by the stack.

>> You have to write correct macros for your microcontroller.

These two expressions are in a logical contradiction.

It seems that these macros do not depend on the driver and will be the same for all drivers.

CANopenNode, 2021-08-20:
CO_CANrxMsg_t is not required by the stack, but macros are. You can write them in different ways.

IhorNehrutsa, 2021-08-22:
So, now, I add both CO_CANrxMsg_t structure and macros to the STM32 CO_driver_target.h
https://github.com/IhorNehrutsa/Intellistep/tree/canopen/src/CANopen/STM32-HAL
I don't know another way.

closed 2023-03-28 by CANopenNode