CANopenNode/CANopenNode #332: WIP: STM32 driver

IhorNehrutsa, 2021-08-20:
I try to use CANopenNode in stepper motor driver https://github.com/IhorNehrutsa/Intellistep/tree/canopen
STM32 driver is here https://github.com/IhorNehrutsa/Intellistep/tree/canopen/src/CANopen/STM32-HAL
CANopen Object Dictionary definition in https://github.com/IhorNehrutsa/Intellistep/tree/canopen/src/CANopen/OD

It based on https://github.com/CANopenNode/CANopenNode/tree/v1.3-master/stack/STM32
and https://github.com/Cyrax86/CanOpenNode-STM32-HAL/tree/master/STM32-HAL

Settings for PlatformIO https://github.com/IhorNehrutsa/Intellistep/blob/canopen/platformio.ini

I am still getting errors
Could anyone help to resolve the errors?

Inviz, 2021-08-20:
I'm also interested in stm32 driver

Inviz, 2021-08-21:
Also i'm wondering where're all the drivers for the up to date canopennode? The drivers I see laying around seem to be for older versions are they? What driver does main developer use?

I'm a novice C/CPP programmer, while I'm pretty good in dynamic languages.  I'd like to maybe give a shot at writing a driver for libopencm3, specifically for stm32/bluepills. I looked at the blank driver and things seem to be quite doable. Eventually I'd love to give freertos integration a shot as well. But currently it's not very clear, would that be redundant?

Inviz, 2021-08-22:
I've started my attempt here https://github.com/Inviz/CANopencm3 but this is just the beginning. Wrote some basic routines for sending and receiving can messages, using interrupts with no hw filters yet. Haven't debugged it either yet

IhorNehrutsa, 2021-08-22:
>Also i'm wondering where're all the drivers for the up to date canopennode? The drivers I see laying around seem to be for older >versions are they? What driver does main developer use?
Answers are here  https://github.com/CANopenNode/CANopenNode/issues/281

Inviz, 2021-08-23:
@IhorNehrutsa Thanks. Yeah, that's what I thought, some fragmentation happened with the new version and drivers haven't caught up just yet. I really like CANOpenNode v4 structure and architecture, I think it's worth writing drivers for it.

What do you use with stm32? I got some progress with the driver, probably this week I'll be able to make the main.c blanks, etc. But it is using libopencm3. What's your framework of choice?

I also plan on adding freertos support, but not sure what's the best way here. Probably will try to combine them both together with defines, instead of trying to maintain two separate drivers.

Inviz, 2021-08-23:
Btw i found this: https://github.com/VanyaTsekhavoi/STM_CAN_DC/tree/797c280b6baba5d5efbafe839d65b45d870105fe/CANopenSTM

CANopenNode, 2021-08-24:
Currently I'm using Linux and PIC32 in my projects. I just don't have time to make another driver(s). I would very like to add driver for other microcontroller, such as stm32.

Inviz, 2021-08-28:
I wrote a driver for flash storage that uses append-only logic for wear-levelling control. It appends entries, until allocated space is full, and then defrags them by erasing all pages and only writing last versions of values. It's not debugged super well just yet, but I'll continue working on it. 

Basically I have can open running on libopencm3 at the basic level. I'll be improving the state of the driver and adding docs during my work. 

I'd like to do freertos integration, but i couldn't find any good examples of main.c structured for freertos. Is anybody familiar with examples of that?

Inviz, 2021-08-29:
Added this freertos blank file. It uses libopencm3, but can be adapted to other stuff. It uses flash storage and the driver from the same project. Persists lss config, etc. Not quite super debugged just yet, but a good start

Inviz, 2021-08-31:
I made freertos use no polling or intervals, besides what is needed. Now FreeRTOS tasks subscribe to CANOpen callbacks to be woken up at the right times. Pretty exciting full package to get anybody started. I added readme for current state and offering of the driver

https://github.com/Inviz/CANopencm3

MaJerle, 2021-09-01:
There is an official STM32 example for CanOpen.

https://github.com/CANopenNode/CanOpenSTM32

Inviz, 2021-09-01:
Ha, that's convenient, an official example since 2 hours ago.  😄  Maybe if we knew a week ago, we would try to write our own drivers. But there're differences too:

It's FD can (an advanced board), HAL/CMSIS, doesn't seem to have a storage driver, and has a pretty simplistic polling rtos integration though. 
Mine is CAN (cheap blue pill), STM32, flash storage, and raw event-driven freertos integration. 

Which are both valid approaches and have different use cases. 

I feel a little competitive but only because I'm such a C novice that it was an accomplishment for me to write mine. But we are in this together, and I'm happy to be around the props. I'm pretty in love with CANopenNode and the ecosystem.

MaJerle, 2021-09-01:
At the moment implementation is simple for everyone to start. I must rework driver to wakeup task when message is received. But indeed "event driven polling" in canopen that needs periodic actions is same as sleeping in task from performance standpoint.

Inviz, 2021-09-01:
> At the moment implementation is simple for everyone to start. I must rework driver to wakeup task when message is received. But indeed "event driven polling" in canopen that needs periodic actions is same as sleeping in task from performance standpoint.

Correct me if I'm wrong, but simply sleeping it task wouldn't allow tickless mode that would put controller to sleep in between timer intervals/messages.

MaJerle, 2021-09-01:
I must correct you :)

Sleep in OS means put task to blocked state and try to run other ones. If no tasks to run, go idle and sleep for "delay milliseconds"

Inviz, 2021-09-01:
Right, but also the CanOpen periodical thread does not *have* to run each 1ms, right? It only does so in polling mode. When driven by specified intervals and events, it may run much less frequently, thus enabling long, predictable stretches of inactivity (i.e. tens of milliseconds to seconds). Thus allowing the OS to optionally put controller to a deep sleep until next interval or interrupt event, which may be better than just idling within OS in regular power mode... At least that's what I gathered?

Anyway, thanks for your replies, just felt like using an opportunity to validate my understanding.

MaJerle, 2021-09-01:
Here you are right. Sleep allows fixed period of idle period, while message queue or semaphore allows longer. But for example, you have to be careful around heartbeat, etc.

A good upgrade of the CanOpenNodw lib would be to tell app max time allowed for sleep before process should be calles next time. This would avoid having timeouts or missed HB intervals

That would be a win-win

Inviz, 2021-09-01:
> A good upgrade of the CanOpenNode lib would be to tell app max time allowed for sleep before process should be calles next time. This would avoid having timeouts or missed HB intervals

Yes, exactly! Glad we are on the same page. The lib already does it though, and it is implemented in my driver. This is the `timerNext_us` argument in all process functions, that can be used to schedule next tick of a thread precisely. But to work properly, that would need subscription to events for interrupts.

CANopenNode, 2021-09-01:
Yes, `timerNext_us` is updated from each CANopen object, including heartbeat, when next interval should occur.

It is funny, I was looking at STM32 for a long time, I even have some unused nucleo boards in my drawer for a few years. Nobody showed interest to publish STM32 driver.
And then you both showed your drivers almost the same day :)
(However, Tilen started, before this issue was opened.)

MaJerle, 2021-09-01:
> > A good upgrade of the CanOpenNode lib would be to tell app max time allowed for sleep before process should be calles next time. This would avoid having timeouts or missed HB intervals
> 
> Yes, exactly! Glad we are on the same page. The lib already does it though, and it is implemented in some of the drivers, like the freertos version of mine. This is the `timerNext_us` argument in all `_process` functions, that can be used to schedule next tick of a thread precisely. But to work properly, that would need subscription to events for interrupts.

Ou, thanks for that. Actually I didnt know about this feature. This helps a lot.

Now, since we know “max” sleep time of the MCU, we can indeed use “osDelay” function that will delay task for the time indicated by timerNext_us parameter. If no other threads are running, FreeRTOS will invoke idle thread and you go to sleep.
If we can properly rely on this variable value, sleep feature and idle time will work perfectly. If in the meantime new message arrives through CAN network, MCU will immediately wakeup.

Planned update of the example includes:
- Ring buffer for received messages
- Messages to be passed to CANopen stack from thread context and not interrupt context
- timerNext_us to be implemented for delay and semaphore wakeup from can thread

I prefer we continue this in CanOpenSTM32 repo instead?

Inviz, 2021-09-01:
I'd like to use this opportunity and ask a relevant question that @CANopenNode can answer, and @MaJerle could potentially benefit from too:

Right now, when CAN interrupt invokes a receive callback, it copies data from message and sets the NEW flag. Later, a process thread can handle that message and clear the flag. If that thread has high priority and executes right after interrupt, it is usually enough and there isn't any buffer needed. Why I like this, is that all messages are grouped by object. So instead of a generic queue that can be filled up by some unimportant CAN message, message arrive into their own individual slots which is wonderful.

However if another CAN message arrives before process happened, often the receive callback discards the message, because it already has one "new" unprocessed message. In this case, i don't understand how to get the feedback from the code that message was discarded. My idea is that I could avoid releasing the CAN message from FIFO queue, if CANOpen couldn't process the message yet, allowing some time to do it later.

My question is: 

1) Is this a realistic situation that messages may be discarded without the bus knowing? Do I need to worry about it? I could imagine that certain situations should not happen at all, like SDO module should not need to be serving multiple clients concurrently, so the conflicting messages shouldn't ever happen. 
2) Do I really need any queue for messages, and if so is there a way to have buffers not mix all messages into one queue? I am afraid that there could be spammy unimportant messages filling up the queue

CANopenNode, 2021-09-01:
There should be no problems with the CANrxNew flag.
For example, SDO in segmented transfer is request-response, in block transfer data is copied to SDO buffer from CANrx interrupt directly. Each SDO channel has own buffer and own CANrxNew flag.
Synchronous RPDO has double receive buffer, etc. But if RPDO is overflowed, it is not a problem in general - latest data is copied to the OD.

Inviz, 2021-09-01:
Right, that's great! So there is no buffer required anywhere at all. Your code is really smartly written and leverages all the canopen wisdom. It is pleasant to work with it and learn too.

MaJerle, 2021-09-02:
Have a look at `develop` branch of CanOpenSTM32 and let me know your thoughts about OS implementation.

https://github.com/CANopenNode/CANopenSTM32/tree/develop

Inviz, 2021-09-02:
This looks pretty good! It would be nice to see putting controller to sleep (which i don't have implemented). What we both don't have implemented either yet is hardware can filters, which seems like a low hanging fruit. But could be really important to avoid too many interrupts.

Since you have good experience with this stuff, I wonder maybe you could provide an example of handling rtos tick time drift more elegantly? RTOS tick precision is quite low (1 tick per 1ms max), while canopennode deals with microseconds. This could potentially lead heartbeats and blinks to drift apart between devices (since it potentially can be mistaken by 0.5ms-1ms each tick). I think we could be able to adjust the time measurements with sysclock data for better precision.

We could only use sysclock for time difference, but then if we use freertos capabilities to disable ticks (tickless mode), we would not be able to enjoy automatic time correction that freertos performs. But then again, does it even matter? Since sysclock can potentially work even while asleep? Please advice.

MaJerle, 2021-09-02:
Sleep is now beyond this example, for the moment at least :)

Hardware filters are not that hard at all, but I wanted to make sure concept works before I go dealing with HW filters.

I'm not sure about time problem, I let you confirm, but is it a problem if there is a drift of 0.5-1ms? That's really low time and considering the fact that CANopen supports 125kHz bitrate, it means in practice MAXIMUM ~1k messages/second, so 1 message/ms. I'm not sure we have an issue here, really. Let me know otherwise indeed

MaJerle, 2021-09-02:
As for splitting things, yes I agree. We need to split things, however first priority is/was to have the things working.

> I know we have these well working clocks in stm32s, we configure it and they can help us. I think in general, maybe using RTOS by itself with 1ms ticks is not a big deal, but getting timeouts through canopennode may result in accumulated time error - i think judging by precision of rtos ticks, it may be possible in theoretical worst case to accumulate 1ms of error per heartbeat. Which could become really noticeable - and it is, in my experiments canopenlinux drifts apart from freertos implementation after a while, judging by candump

This should not be an issue, since every time we execute `process` function, we get a new value for max delay we can afford. I'm really not scary of that at the moment. Important is that CanOpen knows actual time difference between 2 calls and 1ms diff is more than OK.

We should focus on something else, not on 1ms time at the moment :)

Inviz, 2021-09-02:
> In general, I agree to split things, however first priority is/was to have the things working. But if we look closely, reset can only happen if main thread sleeps (main thread considered here a while loop with CO_process), so they never work in parallel. Do we really need to split in this use case? What do we gain by doing this? I don't see performance improvement, but i do see higher memory consumption and more threads in the system (in theory longer context switch).

Yeah you are right. There's no benefit except code structure. Maybe I am thinking about it i a bit incorrectly in that regard.

> This should not be an issue, since every time we execute process function, we get a new value for max delay we can afford. I'm really not scary of that at the moment. Important is that CanOpen knows actual time difference between 2 calls and 1ms diff is more than OK.

> We should focus on something else, not on 1ms time at the moment :)

Alright, it makes sense. First things first, I guess. I brought this up because I was thinking about it, and am curious about what is the right approach. I'll keep an eye on your and other drivers to maybe get some insights.

MaJerle, 2021-09-02:
There have been further updates on the driver done today. That's as best as we can get for the moment.
I try to make it very generic but it won't work, some things (like HAL handles + interrupts) have to be hardcoded, there is no other way.

closed 2023-03-28 by CANopenNode