CANopenNode/CANopenNode #334: Add custom datatype in CANOpen Editor

NebulaDev16, 2021-08-24:
In CANOpenEditor, we can see pre-defined datatypes for CANOpenNode.

[image]

We were exploring CANOpen Editor to add custom datatypes on top of the regular ones. e.g.
TEST16  uint16_t  //2 bytes
TEST32  uint32_t  //4 bytes

Can we do that using CANOpenEditor ?  Or do we need to modify generated OD.c/OD.h /any other file with new datatypes?

Does CANOpenNode support bitfield in OD Entry fields?

NebulaDev16, 2021-09-01:
Thanks for sharing the link. 
As I can check Object dictionary files (OD.c & OD.h) for OD entry 0x2120 - U64 variable at subindex 2, 

1.  datatype is defined in OD entry structure on OD.h
  struct {
        uint8_t highestSub_indexSupported;
        int64_t I64;
        ### uint64_t U64;
        float32_t R32;
        float64_t R64;
        uint16_t parameterWithDefaultValue;
    } x2120_demoRecord;

2. OD entry has dataLength attribute OD.c .  

   .o_2120_demoRecord = {
        {
            .dataOrig = &OD_PERSIST_APP.x2120_demoRecord.U64,
            .subIndex = 2,
            .attribute = ODA_SDO_RW | ODA_TRPDO | ODA_MB,
         ###    .dataLength = 8
        },

So as an experiment, we changed datatype to uint16_t and dataLength to 2. 
We could compile and bring up the application with the changes and U64 read is showing 2 bytes data.
cocomm "4 r 0x2120 2"
[1] 34 12

So for handling custom datatype based data, can we use pre-defined datatype having same memory requirement (2 bytes - uint16_t) and take care of rest in application layer?

CANopenNode, 2021-09-01:
If you are careful, you can also manually edit OD.h/c files. In your case note also default value: `.U64 = 0x1234567890ABCDEF,`

OD interface is not so complex. See function OD_getSub:
https://github.com/CANopenNode/CANopenNode/blob/master/301/CO_ODinterface.c#L182

closed 2022-03-09 by NebulaDev16