CANopenNode/CANopenNode #338: CANopen monitor

IhorNehrutsa, 2021-09-03:
@CANopenNode @MaJerle @Inviz and all
Which CANopen (or CAN bus) monitor do you use for debug/monitor the bus?
Windows applications are preferred.
Which hardware converters (aka CAN2USB) do you use?
Please tell about your work environment (hardware and software).
Thank you.

Inviz, 2021-09-03:
I use linux candump, and cannodelinux with gateway for testing things out. I bought can-usb adapter but it seemed to have died pretty quickly - not sure what happened there

martinwag, 2021-09-06:
Software
- standard linux and stack tools (those run on embedded controller...)
- https://www.uv-software.de/dokuwiki/doku.php?id=uvs:programs:copbrowser_files (GUI Value monitoring)
- Emtas Device Explorer (very powerful, does everything the others do, needs license) 
Hardware
- Peak PCI / USB bus adapter
- Peak PCAN Diag FD Handheld device for field analysis
- Oscilloscope with CAN decoder

closed 2023-03-28 by CANopenNode