CANopenNode/CANopenNode #339: Heartbeat consumer NMT State change

NebulaDev16, 2021-09-07:
We were trying to explore more on NMT Heartbeat producer /consumer functionality.
For this, we have setup two demo devices:
1. HB Consumer - Node 2 -  OD index 0x1016 (subindex 2) as 0x00044e20 - (monitor node 4, consumer HB monitoring time - 20 sec)
2. HB Producer - Node 4 - OD Index 0x1017 (subindex0) as 10000 - (produce HB time - 10 sec)

After this configuration, we can see state change message in node 2 logs:
[image]

But when we reset the node 4 using "cocomm "4 reset node", monitoring node generates emergency message and moves into pre-operational state.

./canopend[16935]: CANopen Emergency message from node 0x02: errorCode=0x8130, errorRegister=0x10, errorBit=0x1B, infoCode=0x00000001
./demoLinuxDevice_addPDO[16948]: CANopen NMT state changed to: "pre-operational" (127)

As we can see,  HeartbeatNmtChangedCallback() only logs the monitored node(4) state change.. How monitoring node(2) state is changed to Pre-operational and EMCY message is generated in this case? Is this application specific implementation or standard mentioned in CiA 301?

NebulaDev16, 2021-09-29:
One more observation to add, when we change state of monitored node(4) from "operational" to "stopped", then node 2 which is monitoring heartbeat of node 4 is remains in operational state.  Tested using cocomm (./cocomm "4 stop"). 

That means Monitoring node (2) is changing its NMT state to pre-operational in case we reset the monitored node (4). In this case. Node 4 state detected in heartbeat consumer callback in node 2 is "unknown(-1)"

_./demoLinuxDevice[4425]: CANopen NMT state changed to: "operational" (5)
./demoLinuxDevice[4425]: CANopen Remote node ID = 0x04 (index = 1): NMT state changed to: "operational" (5)
./demoLinuxDevice[4425]: CANopen Remote node ID = 0x04 (index = 1): NMT state changed to: "stopped" (4)
./demoLinuxDevice[4425]: CANopen Remote node ID = 0x04 (index = 1): NMT state changed to: "unknown" (-1)
./demoLinuxDevice[4425]: CANopen NMT state changed to: "pre-operational" (127)_

JuPrgn, 2021-09-29:
Hi,

You can configure `0x1029 Error behavior object` to change NMT states on specific error types.

When HB is monitored, your monitoring device will send an EMCY. When this happen you probably want to add a callback to stop a motor for example in case the disconnected device is the device that send you commands...

If your monitoring device is alone on the CAN bus, it will not receive CAN message ACK and will send EMCY Tx bus passive, Tx overflow... and will probably switch to PreOperational mode even if you configured 0x1029 to not change NMT.

closed 2022-03-09 by NebulaDev16