CANopenNode/CANopenNode #342: CanOpen FD

tuttoaldoc, 2021-09-16:
Hi,
I'm a user of CanOpenNode since more than 10 years. Is there any plan to upgrade it to support Can FD? 
thanks
Michele

martinwag, 2021-09-16:
#172

tuttoaldoc, 2021-09-16:
Thanks a lot, I didn't see the issue, sorry. I guess there is no update on the topic. Today all the microcontrollers coming out are supporting CAN FD so I guess it's more and more a reality.
thanks
Michele

doughines, 2023-01-20:
Microprocessors have CAN-FD and SocketCAN has some support.

martinwag, 2023-01-23:
socketCAN should fully support FD. CANopenNode lives from users implementing features, so it seems none of them (including me) needed CANopen FD so far. And there is already CAN XL on the horizon...

CANopenNode, 2023-01-25:
I also didn't need CAN FD yet. But it would be nice to use it for example if there is some more process data to send. Now it works good enough with multiple PDOs from one device.

Latest CANopenNode implementation does not limit PDO data to 8 bytes. So basically "only" driver supporting CAN FD is required...

tuttoaldoc, 2023-01-25:
> I also didn't need CAN FD yet. But it would be nice to use it for example if there is some more process data to send. Now it works good enough with multiple PDOs from one device.
>
> Latest CANopenNode [...]

Hello,
what about SDO? Is it also possible to set a bigger SDO data size?
thanks
michele

martinwag, 2023-01-25:
You need to implement USDO part for that.