CANopenNode/CANopenNode #346: Question: Where do you save the bit timing parameter (baud rate) in the Object Dictionary?

DanielMartensson, 2021-10-17:
Hi!

I'm building a CANopen library at my GitHub and I want to ask a simple question.
Where do you save the bit timing parameter (baud rate) in the Object Dictionary?

Is it inside communication parameter area (Index: 0x1000 – 0x1FFF) or manufacturer parameter area (Index: 0x2000 – 0x5FFF)?
I.m.o it would be good if it's inside the communication parameter area because bit timing is related to communication.

DanielMartensson, 2021-10-18:
> Bit timing is not part of the standard CiA301 OD. You can implement it somewhere in the manufacturer area if you need it.

Thank you. Same as Node ID? Beacuse I don't see Node ID is saved inside the communication parameter area.

martinwag, 2021-10-18:
Yes. 

Get yourself a copy of CiA301 from CAN in Automation, it's free of charge.

martinwag, 2021-10-18:
From what I can see on a short look, the nanotec link seems to be OK for basic understanding of functionallity. I'm not going to check the whole page / your 2002 305 against current specifications. CANopenNode should meet current 305 (3.0.0 / 2013) in this regard (I wrote the LSS part).

You've mentioned in your first post that you want to build your own CANopen Library. If you are **serious** about this you will need full access to current CiA documents, which means becoming a member...

DanielMartensson, 2021-10-18:
> ?? Message 0x7E4 is LSS slave resposne message, this is correct.
> 
> If you want to do this for hobby / educational purpose, I can only reccomend you to get a larger micro (ESP32 has CAN bus, or STM32, ...).

Yes, the address is correct, but the response is wrong. I have looked it up. 

Anyway, I'm already using STM32 with my SAE J1939 library available on my GitHub.