CANopenNode/CANopenNode #348: node guarding

bbhk17, 2021-10-19:
Hi Guys,

First thanks for the work seams to work pretty well on my hardware !

I have a question, I have to use node guarding instead of heart beat, any idea about how to do/setup this request ?

Thx !

qwerty258, 2021-11-24:
[CiA 802 application note](http://www.can-cia.org/standardization/technical-documents/)
[CiA error protocols](https://www.can-cia.org/can-knowledge/canopen/error-control-protocols/)
> Do not use Node/Life-guarding
> The Node/Life-guarding function is based on CAN remote frames. For many years, CiA has not recommend the use of CAN remote frames. They make more problems than they solve. This is why CiA recommends using the Heartbeat function instead in order to confirm the NMT command and to double-check that the node is still available.

closed 2023-03-28 by CANopenNode

chuxinyiyou, 2023-05-12:
> 嗨，伙计们，
> 
> 首先感谢工作接缝在我的硬件上运行良好！
> 
> 我有一个问题，我必须使用节点保护而不是心跳，关于如何做/设置此请求的任何想法？
> 
> 感谢！
Hey buddy, have you implemented this feature yet，Node protection