CANopenNode/CANopenNode #362: Should heartbeat timeouts occur even for nodes that have not been active before?

cmorganBE, 2021-11-05:
CO_HBconsumer_process() has this check:

```
                CO_FLAG_CLEAR(monitoredNode->CANrxNew);
            }

            /* Verify timeout */
            if (monitoredNode->HBstate == CO_HBconsumer_ACTIVE) {     <-------------------------------------------
                monitoredNode->timeoutTimer += timeDifference_us_copy;

                if (monitoredNode->timeoutTimer >= monitoredNode->time_us) {
                    /* timeout expired */
#if (CO_CONFIG_HB_CONS) & CO_CONFIG_HB_CONS_CALLBACK_MULTI
                    if (monitoredNode->pFunctSignalTimeout!=NULL) {
                        monitoredNode->pFunctSignalTimeout(
                            monitoredNode->nodeId, i,
                            monitoredNode->functSignalObjectTimeout);
                    }
```

Shouldn't this check be:

```
if(((monitoredNode->HBstate != CO_HBconsumer_UNCONFIGURED) && (monitoredNode->HBState != CO_HBconsumer_TIMEOUT)) {
```

to allow heartbeat timeouts to occur on nodes that have been configured for heartbeats but for whom heartbeats have not been received?

We are resetting and configuring devices based on heartbeat timeouts and for a device that never responds I'd like to use heartbeat timeouts to handle the case where we reset a device but never hear from it, unless there is another better mechanism to use and the change above isn't valid.

Regards,
Chris

martinwag, 2021-11-05:
was changed in commit c19879b659ac8ee1cdd987391c04ce42d92dcbe1

before that, it was 

```
                if(monitoredNode->HBstate!=CO_HBconsumer_UNCONFIGURED &&
                   monitoredNode->HBstate!=CO_HBconsumer_UNKNOWN) {
```

Unfortunately, commit message does not give a hint for why that was changed.

martinwag, 2021-11-05:
Not triggering heartbeat timeout before a device is seen for the first time is intentional. If you do a reset on your device, it should send the power up message and start sending heartbeats. You can set a callback for first received heartbeat if you need that info.
This is designed so that you can just start all 127 heartbeat consumers, and when a device is added you get notified and heartbeat monitoring is startet automatically.

cmorganBE, 2021-11-05:
@martinwag the use case is that I'd like CON to send hb timeouts rather than my app keeping track of time, as CON Is already keeping track of timeouts. I'm new to CON but have developed a number of libraries, OSS and otherwise, and this feels like a reasonable thing for the stack to help a little more with.

If I change as above check I get a timeout when the device isn't present as expected.

I did have to make one other set of changes that I perform each time I reset the device so I can get more HB timeouts. Otherwise the HBstate change to TIMEDOUT locks out future callbacks, and also the timeoutTimer should probably be reset.

```
         // in my code somewhere (I was thinking this could possibly be a new method in CO_HBconsumer.h/.c)
        hbNode->HBstate = CO_HBconsumer_UNKNOWN;
        hbNode->timeoutTimer = 0; // zero the time since we just reset the heartbeat

        CO_NMT_SendCommand(instance, CO_NMT_RESET_COMMUNICATION, nodeId);
```

but the above code needed one more tweak, updating the monitoredNode prior to the callback, otherwise if I reset the HBstate and timeoutTime in the callback itself they get overwritten when the callback returns:

```
                if (monitoredNode->timeoutTimer >= monitoredNode->time_us) {
                    /* timeout expired */
                    CO_errorReport(HBcons->em, CO_EM_HEARTBEAT_CONSUMER,
                                   CO_EMC_HEARTBEAT, i);
                    monitoredNode->NMTstate = CO_NMT_UNKNOWN;                <-------------------- these are ahead of cb now
                    monitoredNode->HBstate = CO_HBconsumer_TIMEOUT;         <----------------------

#if (CO_CONFIG_HB_CONS) & CO_CONFIG_HB_CONS_CALLBACK_MULTI
                    if (monitoredNode->pFunctSignalTimeout!=NULL) {
                        monitoredNode->pFunctSignalTimeout(
                            monitoredNode->nodeId, i,
                            monitoredNode->functSignalObjectTimeout);
                    }
#endif
                }
```

cmorganBE, 2021-11-05:
@martinwag would it help if I opened up a PR so you can see what I'm proposing?

cmorganBE, 2021-11-05:
Here is the pr, https://github.com/CANopenNode/CANopenNode/pull/363

trojanobelix, 2023-05-22:
DS301 stated that 

_Monitoring of the heartbeat producer shall start after the reception of the **first** heartbeat._

So it is not allowed that heartbeat timeouts occur on nodes that have been configured for heartbeats but for whom heartbeats have not been received.

cmorganBE, 2023-05-22:
ahh k. I figured there was some spec reason for it.

trojanobelix, 2023-05-25:
Sorry, but the standard ist a standard, and CANopenNode should be close to it.
Please close this issue, if no questions left. And also the PR. Janez seems to be busy and I do not have the privilege to close. I maintain the Editor.

closed 2023-05-26 by cmorganBE