CANopenNode/CANopenNode #366: Node should be set to Pre-Op, when Boot-up is received

gseqBE, 2021-11-11:
Hello,

We have a setup running CANOpenNode on a Master and a COTS  slave communicating with it.

Looking at [line 50](https://github.com/CANopenNode/CANopenNode/blob/master/301/CO_HBconsumer.c#L50) in file co_HBConsumer.c, NMTState is based on the content of the NMT message. 

In the case of a bootup NMT message, this transitions NMTState to Initializing, but we would expect, based on CANopen spec below for the device to be in pre-operational. 

From the CiA Specification(CiA 301 version 4.2.0, CANopen application layer and communication profile):
Section: 7.2.8.2.3.1

> Through this service, the NMT slave indicates that a local state transition **occurred** from the NMT state  Initialisation to the NMT state Pre-operational.

This means the node successfully went to pre-op when the boot-up message was sent.

1.  Could this be considered as a deviation from the CANOpen spec? Shouldn't the stack mark the node as pre-op when a boot up message is received?

2. The CANOpen spec does **not** specify that all slaves should generate heartbeats. What happens when such a device is on the bus.? Currently the stack marks the device as in "Initializing State", when actually it has already transitioned to "Pre-Op", but does not send out heartbeats.

gseqBE, 2021-11-18:
We would appreciate any comment on this.

martinwag, 2021-11-19:
Hi,

go to figure 46 in 301, this defines the bootup message as message with payload "0". So the implementation should be according to spec.

cmorganBE, 2021-12-09:
@martinwag I think the question is whether that value might be miss named, for example:

https://workarea.ego-gw.it/ego2/ego/itf/software/301_canopen.pdf

From the bottom of page 71:

```
r: reserved (always 0)
s: the state of the Heartbeat producer
0: BOOTUP
4: STOPPED
5: OPERATIONAL
127: PRE-OPERATIONAL
```

This would indicate that perhaps we should change:

```
typedef enum {
    /** -1, Device state is unknown (for heartbeat consumer) */
    CO_NMT_UNKNOWN = -1,
    /** 0, Device is initializing */
    CO_NMT_INITIALIZING = 0,
    /** 127, Device is in pre-operational state */
    CO_NMT_PRE_OPERATIONAL = 127,
    /** 5, Device is in operational state */
    CO_NMT_OPERATIONAL = 5,
    /** 4, Device is stopped */
    CO_NMT_STOPPED = 4
} CO_NMT_internalState_t;
```

to

```
typedef enum {
    /** -1, Device state is unknown (for heartbeat consumer) */
    CO_NMT_UNKNOWN = -1,
    /** 0, Device is initializing */
    CO_NMT_BOOTUP = 0,   <-------------------------------------------------
    /** 127, Device is in pre-operational state */
    CO_NMT_PRE_OPERATIONAL = 127,
    /** 5, Device is in operational state */
    CO_NMT_OPERATIONAL = 5,
    /** 4, Device is stopped */
    CO_NMT_STOPPED = 4
} CO_NMT_internalState_t;
```

Except the next section on page 72 has this text:

```
9.2.6.2.3 Bootup Protocol
This protocol is used to signal that a NMT slave has entered the node state PRE-OPERATIONAL after
the state INITIALISING. The protocol uses the same identifier as the error control protocols.
```

So now I suspect I don't understand the various states and one thing has a BOOTUP state and another an INITIALIZING state....

cmorganBE, 2021-12-09:
Yeah page 78 makes it more clear and shows INITIALIZING, PRE-OP, OP, and STOPPED.  So nevermind my above comment.

cmorganBE, 2021-12-13:
@martinwag alright, I spoke with Gerard and I was unclear about what he was proposing.

Looking at the CANopen spec it looks like CON should be considering a value of 0 (bootup state), to be preoperational, as a bootup message is only sent/received when the node is transitioning to preoperational.

Eg. we should modify:

```
typedef enum {
    /** -1, Device state is unknown (for heartbeat consumer) */
    CO_NMT_UNKNOWN = -1,
    /** 0, Device is initializing */
    CO_NMT_INITIALIZING = 0,
    /** 127, Device is in pre-operational state */
    CO_NMT_PRE_OPERATIONAL = 127,
    /** 5, Device is in operational state */
    CO_NMT_OPERATIONAL = 5,
    /** 4, Device is stopped */
    CO_NMT_STOPPED = 4
} CO_NMT_internalState_t;
```

to be 
```
typedef enum {
    /** -1, Device state is unknown (for heartbeat consumer) */
    CO_NMT_UNKNOWN = -1,
    /** 127, Device is initializing */
    CO_NMT_INITIALIZING = 127,  <----------------------------
    /** 0, Device is in pre-operational state */
    CO_NMT_PRE_OPERATIONAL = 0, <----------------------
    /** 5, Device is in operational state */
    CO_NMT_OPERATIONAL = 5,
    /** 4, Device is stopped */
    CO_NMT_STOPPED = 4
} CO_NMT_internalState_t;
```

CANopenNode, 2023-03-28:
Trying to go through old issues...
What if device enters operational directly after bootup?

closed 2023-04-11 by CANopenNode

AikiWolf, 2024-01-03:
Hey all!

So per CiA 301, if following the NMT state machine to the letter, a slave device will send one bootup message (700h+Node ID) with a data load of 1 byte containing 00h. This is to indicate that it has "entered the bus" so to speak. 

Once it has accomplished it's internal bootup and is ready to receive messages, it will automatically send another message of the same CAN-ID (700h + Node ID), with a payload of 1 byte containing 7Fh. This is all without any external influence and is called the bootup protocol, and is mandatory for all CANOpen devices regardless of whether heartbeat is implemented. The general confusion here is the message sent as the bootup protocol is the same message sent at intervals when the heartbeat is implemented. 

If the device is set to produce a heartbeat, it will send send this same message with the same payload at an interval defined by the Heartbeat Producer Time object dictionary entry. If it is not set to produce a heartbeat, it will do nothing else.

The NMT master, upon receipt of the initial "Pre-operational" message sent as part of the bootup protocol and assuming its own initialization and transition through internal mechanisms to the "start" state, will send an NMT start command (CAN-ID 00h, payload 01h [NODE ID of target node]h ). 

So for example the NMT master would send, to a NMT slave with NODE ID 02h: COB-ID: 00h Data Byte 0: 01h Data Byte 1: 02h

The NMT slave would receive that command and AT THE NEXT HEARTBEAT INTERVAL, send COB-ID: 702h Data Byte 0: 05h. 

This indicates it has moved into NMT 'operational' state.

Hope this helps!

CANopenNode, 2024-01-03:
> Once it has accomplished it's internal bootup and is ready to receive messages, it will automatically send another message of the same CAN-ID (700h + Node ID), with a payload of 1 byte containing 7Fh.

This message is not part of a bootup. It is the first heartbeat message and is send only, if object 0x1017 (heartbeat interval) is different than zero.

After a bootup message (700h+Node ID) with a data load of 1 byte containing 00h, CANopen device is actually ready. NMT master may react also on this message.

Device may enter operational on its own and this may be indicated by the first heartbeat message (Data Byte 0: 05h). (And this first heartbeat very soon follows the bootup message in CANopenNode)

After that some other device, which monitors heartbeat from this (and other) nodes, knows, that its network is fully operational.