CANopenNode/CANopenNode #369: Packet missed between main and RT thread event notification

NebulaDev16, 2021-11-23:
We are planning to use 2 thread model(Main and RT Thread) of CANOpenNode. As we can see, Main thread runs on 100 millisec and RT thread runs on periodicity of 1 millisec seconds.

We can see RT thread updates the 
1. global structure mapped to RX Array (separate for HB Consumer, SDO Client) (which is read by MAIN Thread later for processing)
2. Event is updated on main fd. (using wakeupcalback() API) 

If multiple message is received by RT thread belonging to one RX Array e.g. SDO Response. 
In this case, does RT thread will overwrite the data on global structure with latest data (and old CAN packet info is lost)? 
Or does RT thread queue two separate events to Main thread (and CAN packet information is not lost)?

Please confirm if the understanding is correct?

martinwag, 2021-11-23:
All services, except for SDO block and Emergency, are request - response, meaning that more than one rx message is a protocol violation. 
- SDO block has a designated, user sized buffer to avoid overwriting
- Emergency uses callback instead of buffer
For the others, if more than one message is received, the old one remains inside the buffer ( see IS_CANrxNew() ) and the new one is dropped.

NebulaDev16, 2021-11-24:
Thanks for your response.

We have one query-
Suppose we have CAN topology of 3 nodes in which 2 nodes acting as CAN SDO Client(Node 1 and Node 2) and 1 acting as SDO Server(Node 3).

Now if the sequence is:
1. SDO Client(Node 1) initiates SDO Write transaction on SDO Server (Node 3).
2. Node 3 starts the processing SDO Client Request(0x603) and CAN Bus is free.
3. SDO Client(Node 2) gets the bus and initiates SDO Write transaction on SDO Server (Node 3).
4. Node 3 drops the SDO Request from Node 2.
5. Node 3 broadcasts SDO response on network (0x583). 
6. Node 2 donot know its packet is drooped. And now Node 1 and Node 2 considers SDO response as valid response(0x583). 

Problem is that in this case, Node 2 SDO request is dropped but still it is assuming that it received the valid response(as Cob-id of SDO response 0x583 is same).

How CAN Open standard will handle this kind of situation in multi-SDO client topology?

martinwag, 2021-11-24:
You're not allowed to use a SDO server with two clients at the same time. 
To solve this, you have two choises:
- Implement one server per client, so two servers in Node 3. They can access the same OD, but all the processing is then available twice. Take a look at OD 0x1200. Be aware that the number of SDO servers in a network is limited. To get this working with CANopenNode, you need to add a second (n-th) SDO server in initializer function.
- Implement CiA 305 SDO manager. You then need to register SDO with the manager before access and release afterwards.

Hope that helps

closed 2021-12-06 by NebulaDev16