CANopenNode/CANopenNode #370: flagPDObyte use as COS for PDOs?

pseudogelb, 2021-11-24:
Hi,

In the old CANopenNode Version it was possible to directly activate COS in the Object Dictionary Editor,
this seems not to be possible anymore in the new version. Now I have seen, that for a TPDO there
is checked, if the flagPDObyte ist set and I think this can be used for COS, but I don't see how this could be done.

How can I send a TPDO only when a value change occurs?

pseudogelb, 2021-11-24:
I found the following:

OD_requestTPDO(OD_getFlagsPDO(OD_ENTRY_H210A),0);

The problem is, the flags are still NULL, so this should be the correct way to trigger a COS, but I don't see how to proper initalize the TPDO for this to work

pseudogelb, 2021-11-24:
After extension initialization it seems to work, but it always sends me a zero value. Should I set the OD variable through a function? When I directly assign the value, there is always zero.

CANopenNode, 2023-03-28:
v1.3 has some automation regarding COS functionality. In v4 PDOs are more flexible and linked to OD variables to which they are mapped. There is no automatic COS detection, application must trigger PDO, indirectly from OD variable. See documentation and https://github.com/CANopenNode/CANopenDemo

Example:
https://github.com/CANopenNode/CANopenDemo/blob/master/demo/CO_application.c#L36