CANopenNode/CANopenNode #374: Heartbeat timeouts causing master to go out of operational

cmorganBE, 2021-12-10:
I believe we've done our due diligence and understand the issue but wanted to get a second set of eyes.

Linux system, CanOpenNode master, several devices on the bus, all of which are configured for heartbeats.

Noticed that sometimes at system startup we weren't getting any RPDOs. Traced this back to our state (as the master) going from operational back into preop and the root cause being that one of the monitored nodes has a heartbeat timeout. This timeout sets an errorBits which CO_EM_process() turns into a value in errorRegister, and which CO_NMT_process() then turns into a preop here:

```
    bool_t errRegMasked = (NMT->NMTcontrol & CO_NMT_ERR_ON_ERR_REG) != 0
                    && (CO_getErrorRegister(NMT->em) & NMT->NMTcontrol) != 0;
```

Because the error register value & with NMT->NMTcontrol != 0:

```

/**
 * Default values for CO_CANopenInit().
 */
#define NMT_CONTROL \
           static_cast<CO_NMT_control_t>( CO_NMT_STARTUP_TO_OPERATIONAL \
          | CO_NMT_ERR_ON_ERR_REG \
          | CO_ERR_REG_GENERIC_ERR \
          | CO_ERR_REG_COMMUNICATION )
```

The solution we have so far:

- We register an Rx emergency message callback
- In the callback we clear the errorBits and clear errorRegister
- This worked BUT because of the inhibit time we have set of 100ms we get the first callback when the emergency is sent, but miss a few as heartbeats result in additional emergency messages during the inhibit time.
- However, even though CO_EM_process() doesn't send the emergency message it does set em->errorRegister, so the next time CO_NMT_process() runs again we end up in preop.
- So we can fix this by reducing the inhibit time to zero but the whole approach feels like we are missing something.

Should we instead be modifying NMT_CONTROL to remove 'CO_ERR_REG_COMMUNICATION'?
- errorRegister has this value OR'ed in in CO_EM_process() when errorBits has any value in byte 2 or 3
```
#define CO_CONFIG_ERR_CONDITION_COMMUNICATION (em->errorStatusBits[2] \
                                            || em->errorStatusBits[3])
```
```
ifdef CO_CONFIG_ERR_CONDITION_TEMPERATURE
    if (CO_CONFIG_ERR_CONDITION_TEMPERATURE)
        errorRegister |= CO_ERR_REG_TEMPERATURE;
#endif
    if (CO_CONFIG_ERR_CONDITION_COMMUNICATION)    /* see above #define */
        errorRegister |= CO_ERR_REG_COMMUNICATION;
#ifdef CO_CONFIG_ERR_CONDITION_DEV_PROFILE
    if (CO_CONFIG_ERR_CONDITION_DEV_PROFILE)
        errorRegister |= CO_ERR_REG_DEV_PROFILE;
#endif
```
- This change would mean we'd miss out on any communication errors

The other thought was to modify NMT_CONTROL to remove 'CO_NMT_ERR_ON_ERR_REG'. This is more of a hammer than a scalpel though, even though the comment kind of aligns with what we want to do here, we don't want a

martinwag, 2021-12-10:
Approach 4: You can configure that behaviour with OD 0x1029 Error behaviour, so no change to source code needed. Setting index 1 to 0x01 should disable pre-op on HB consumer error.

If staying in operational when one of your nodes is missing is OK depends on your appplication :-)

cmorganBE, 2021-12-10:
Hi @martinwag, approach 4 looks ideal but doesn't appear to be working.

How does that value propagate or is used by the stack? I searched for all instances of 1029, 'errorBehavior' (case insensitive), and any variables set to 1029 and didn't spot anything.

It looks like NMT->NMTControl is the mask in CO_NMT_process() for detecting the error and setting errRegMasked to true. And NMTControl is passed down from CO_CANopenInit() or is there some other place 

Are we missing something?

martinwag, 2021-12-13:
Search for "OD_errorBehavior" register. That register is passed by main to NMT / Heartbeat handler. 

This did work in v1.3, I don't know about current version.

LluisTrillaClaitec, 2022-05-12:
Seems like it was removed in preparation for CANOpen v2 (does the standard not support it or was it just left out?) as per commit f73025ef651ebc62d78a670c54b66f4f18bde84b

CANopenNode, 2023-03-28:
I didn't answer completelly. Redefine CO_CONFIG_ERR_CONDITION_COMMUNICATION condition to anything you like (use mask to disable, CO_driver_custom.h file):
https://github.com/CANopenNode/CANopenNode/blob/master/301/CO_config.h#L280

closed 2023-04-11 by CANopenNode