CANopenNode/CANopenNode #375: Application RPDO Callback example needed

c031917, 2021-12-21:
First: Brilliant work! Really a gem in the Open Source world.
I want to use your code on STM32 to read out and plot encoder data. I think I understood the most steps how to make the solution. But I still struggle not with the "last mile", but with the "first mile" - how to integrate the stack into a wider application perspective. Links to the docs I found were related to the Linux implementation or to older versions. Whole embedded application examples are hard to find. One V4 example I found is [https://github.com/Jexbat/CANopenSTM32-HAL], where at least there is a button and four LEDs coupled to the OD. The author added:
In CO_storage.h
```
typedef struct {
    OD_extension_t OD_1010_extension; /**< Extension for OD object */
    OD_extension_t OD_1011_extension; /**< Extension for OD object */
    **OD_extension_t OD_6200_extension; /**< Extension for OD object */**
```
and
```
CO_ReturnError_t CO_storage_init(CO_storage_t *storage,
                                 CO_CANmodule_t *CANmodule,
                                 OD_entry_t *OD_1010_StoreParameters,
                                 OD_entry_t *OD_1011_RestoreDefaultParameters,
				 OD_entry_t *OD_6200_LED,
```
In CO_storage.c:
...
```
static ODR_t OD_write_6200(OD_stream_t *stream, const void *buf,
                           OD_size_t count, OD_size_t *countWritten)
{
    /* verify arguments */
    if (stream == NULL || stream->subIndex == 0 || buf == NULL
        || countWritten == NULL
    ) {
        return ODR_DEV_INCOMPAT;
    }
    ODR_t returnCode = ODR_OK;
    uint8_t data = *((uint8_t *)buf);
    switch(stream->subIndex){
    case 1:
        (data > 0)?HAL_GPIO_WritePin(D1_GPIO_Port, D1_Pin, GPIO_PIN_RESET):HAL_GPIO_WritePin(D1_GPIO_Port, D1_Pin, GPIO_PIN_SET);
        break;
    case 2:
        (data > 0)?HAL_GPIO_WritePin(D2_GPIO_Port, D2_Pin, GPIO_PIN_RESET):HAL_GPIO_WritePin(D2_GPIO_Port, D2_Pin, GPIO_PIN_SET);
        break;
    case 3:
        (data > 0)?HAL_GPIO_WritePin(D3_GPIO_Port, D3_Pin, GPIO_PIN_RESET):HAL_GPIO_WritePin(D3_GPIO_Port, D3_Pin, GPIO_PIN_SET);
        break;
    case 4:
        (data > 0)?HAL_GPIO_WritePin(D4_GPIO_Port, D4_Pin, GPIO_PIN_RESET):HAL_GPIO_WritePin(D4_GPIO_Port, D4_Pin, GPIO_PIN_SET);
        break;
    }

    if (returnCode == ODR_OK) *countWritten = sizeof(uint8_t);
    return returnCode;
}
```
...
```
CO_ReturnError_t CO_storage_init(CO_storage_t *storage,
                                 CO_CANmodule_t *CANmodule,
                                 OD_entry_t *OD_1010_StoreParameters,
                                 OD_entry_t *OD_1011_RestoreDefaultParameters,
				OD_entry_t *OD_6200_LED,
```
...
```
    if(OD_6200_LED != NULL){
        storage->OD_6200_extension.object = storage;
        storage->OD_6200_extension.read = OD_readOriginal;
        storage->OD_6200_extension.write = OD_write_6200;
        OD_extension_init(OD_6200_LED,
           storage->OD_6200_extension);
    }
```

Is this really the way to go ?  I feels wrong to "patch" so many files just to link application code to CAN objects. I have seen the CO_application.c file (no CO_application.h file by the way) -  is this a hint that you propose putting the linking code here ?
What I'd want is a full example with in/out interface from an application, based on Best Practice you have in mind. Probably for you this is a no-brainer, for us as _consumers_ it is pain to find it out by reverse-engineering, especially as things changed a lot over the versions.
A simple "steps to use it in your application" would also help. Besided the obvious steps that you have to prepare OD.c and OD.h via .eds editor, something like:
* add the callback code to execute when object changes, to CO_storage.c/h (or CO_application.c/h?)
* add blabla to your main.c user code loop
* don't change x,y,z files
 
Coming back to my overall question: Is there an example for an MCU application on V4.0  using callback available ?

martinwag, 2021-12-22:
There are some example for CANopenNode here: https://github.com/CANopenNode/CANopenPIC . I assume you already found those, they are also not fully working devicves.

- Callback should work like this, at least in V1.3. I don't think that has change.

```
CO_SDO_abortCode_t temperature_callback(CO_ODF_arg_t *p_odf_arg) {};

CO_OD_configure(CO->SDO[0], OD_2108_temperature, temperature_callback, this, NULL, 0);
```
You can put this anywhere in your code. The same goes for direct OD access, just use the macros provided by OD generator. There are also methods available for accessing OD using entry numbers ("0x1111 12") instead of direct access.

- I don't think you need to modify the storage stuff if you're happy with just generic store/restore all variables. You for sure don't need to set LEDs inside that code.

c031917, 2021-12-22:
ok, you are confirming at least that the code I found is sub-optimal.
I tried to introduce your proposal and I get  on compile "unknown type name 'CO_ODF_arg_t'"  .  In the CANopenPIC code this type does even not exist - so no help. What I tried:

```
#include "main.h"
#include "CANopen.h"
#include "OD.h"

CAN_HandleTypeDef hcan;

void SystemClock_Config(void);
static void MX_GPIO_Init(void);
static void MX_CAN_Init(void);

CO_SDO_abortCode_t temperature_callback(CO_ODF_arg_t *p_odf_arg)
{
    /* here we will put the code to be executed when a PDO arrived */
};

int main(void)
{
    CO_OD_configure(CO->SDO[0], OD_2108_temperature, temperature_callback, this, NULL, 0);
}

```

What's wrong ?

martinwag, 2021-12-22:
As said, this is for v1.3. In that, function comment on CO_OD_configure() explains how to use it and which types you need. This or something similar should exist in v4 also. I have not used v4, so I can't help you there.

koson, 2022-01-06:
Found CO_ODF_arg_t  at https://github.com/hwidvorakinfo/F4_CANopen/blob/5247dd9523cf5a6c1a7bdfab8c2446b7f9bc5ee8/canopennode/stack/CO_SDO.h

Maybe help.

GiulioDallaVecchia, 2022-06-20:
Hi,

is there any possibility to have a callback from stack when a RPDO is processed?

GiulioDallaVecchia, 2022-06-23:
I've just discovered the solution. It's just enough to use `OD_extension_init `before to call `CO_CANopenInitPDO`:

```
static OD_extension_t TempData_extension;
TempData_extension.object = me;
TempData_extension.write = OD_write_temperatureData;
TempData_extension.read = NULL;
OD_extension_init(OD_find(OD, 0x2001), &TempData_extension);

/* Initialize PDO */
err = CO_CANopenInitPDO(me->CO, me->CO->em, OD, activeNodeId, &errInfo);
```

CANopenNode, 2022-06-24:
Yes, it is something like that.
It is not practical to have direct access to PDO, because of PDO mapping. However, when PDO reads or writes OD variable, it uses its read/write function...

Rukakuka, 2022-08-08:
> Yes, this is the solution. Does it suit your need? Thing is, application doesn't know, how PDOs will be configured. Also application doesn't need to worry about mapping. Just use callback on OD object and it should work.

Could you please elaborate on this?
I'm doing the same as @GiulioDallaVecchia in the [solution](https://github.com/CANopenNode/CANopenNode/issues/375#issuecomment-1164085838) but my function seems to fail to bind. I use CanOpenLinux example, so I follow the initialization process in the `CO_main_basic`.

I have in my `CO_main_basic`.c`
```
static OD_extension_t OD_2110_1_custom_extension;
```
Then, I call `OD_extension_init` before the `CO_CANopenInitPDO`.
```
D_2110_1_custom_extension.object = &myObject;
OD_2110_1_custom_extension.read = OD_readOriginal;
OD_2110_1_custom_extension.write = OD_writeOriginalWithCallback;
OD_extension_init(entry, &OD_2110_1_custom_extension);
```
where `OD_writeOriginalWithCallback` is declared in the `CO_ODinterface.h` and defined in `.c`
```
ODR_t OD_writeOriginalWithCallback(OD_stream_t *stream, const void *buf, 
OD_size_t count, OD_size_t *countWritten)
{
    __log_printf(0, "OD write with callback");
    // Write to object as usual
    ODR_t ret = OD_writeOriginal(stream, buf, count, countWritten);
    // Callback function - todo
    return ret;
}
```

But when I get PDO message from the bus (from another node I configured), the function `OD_writeOriginal` is called, I can check it with similar `log` statement in it, which I added into it.
 So, I can't figure out what I mess with `extenson_init`, could you please help? To add, I checked function pointer of `OD_writeOriginalWithCallback` before and after `CO_CANopenInitPDO` and they are different. How to properly bind my function to the OD write function through `extension`?