CANopenNode/CANopenNode #377: Main_blank ERROR

CarmenMFM, 2022-02-10:
Hello!
I have modified OD.h and OD.c because i have errors when I compile with sudo make in Linux terminal.
When i resolved these errors and I recompile with sudo make, i have a new error with main_blank. This error says that reference to `OD_PERSIST_COMM' undefined but this reference is defined in OD.h.
Next I show the error and the definition: 

[image]

[image]

Can someone please help me? Thanks!

CarmenMFM, 2022-02-10:
@martinwag Thanks!
Now i have a new problem, when i read a parameter from my object dictionary in Linux terminal appear the next error:
sudo cocomm "10 r 0x6070 0 u16"
[1] ERROR:0x06020000 #Object does not exist.
I don now what parameter must appear in the struct OD_PERSIST_COMM_t.
Next, i leave my OD.h 
If you can, could you see this file?
Thanks again!
[OD.txt](https://github.com/CANopenNode/CANopenNode/files/8040462/OD.txt)

CarmenMFM, 2022-02-10:
Also,
the struct OD_PERSIST_COMM_t. didn't appear in OD.h and I include this struct how appear in the od.h of example.
I soppose that data is missing but i don know which ones.

martinwag, 2022-02-10:
Your'e not supposed to edit OD files manually. use this software: https://github.com/CANopenNode/CANopenEditor. Load the current file, do an export, then all this stuff should be set for you.

CarmenMFM, 2022-02-10:
I know...
I have used this application and the dictionary that it has generated gave errors, that is why I had to modify it.

CarmenMFM, 2022-02-11:
Hello, sorry to bother again.
I have managed not to have to manually edit OD.h and OD.c but I still get the error: "  #Object does not exist."
I'm not sure if the error is because the size of the object dictionary is not updated correctly, should I specify it somewhere?
Thanks!

CANopenNode, 2023-03-28:
If this is still actual, please re-open the issue.