CANopenNode/CANopenNode #381: emergency message

franck48, 2022-03-08:
hello ,
i need a little help,

i use CanopenNode on a Microchip 16bits Dsp ds33E,
sometimes, once very 1 or 2 days my device stop to work, return to pré-op state
and send an emergency message 

0x0B2  00 61 01 29

the node own the ID  32 so 0x80 + 0x32.

the ERROR CODE seems to be internal error
#define CO_EMC_SOFTWARE_INTERNAL        0x6100U /**< 0x61xx, Internal Software */
and the STATUS BITS indicate
#define CO_EM_CAN_BUS_WARNING           0x01U /**< 0x01, communication, info, CAN bus warning limit reached */

very difficult to try to produce it whan i want,
what dos mean :   CAN bus warning limit reached    ?

JuPrgn, 2022-03-08:
Hi Franck

CAN bus warning appears when your error counters are above a limit this is detailed in CAN bus protocol error handling (not CANopen). This mean that you have some error on your frames (issues with frequency, bit timing, CAN bus length...) or this can happen if your node is momentary "alone" on a bus as his messages will not be acknowledged properly and then increase error counters.

franck48, 2022-03-08:
hello JuPrgn,
i just update my message with more informations,
i do a mistake in my orginal message, the probleme seems to be around the IT-TIMER  1ms

JuPrgn, 2022-03-08:
Given your description I guess your timer interrupt is sometimes interrupted by another one that block longer than 1ms and this occur every 1 or 2 days. You are right this is hard to debug... 
Can you change your other interrupts priority to be loyer than this timer ?  (CAN interrupt should still be higher priority than CAN Timer)

franck48, 2022-03-08:
humm nice advice, once more mr thank you for your help,
i try it immediately.
i will post the result here

JuPrgn, 2022-03-08:
You are welcome.
Hope this will helps you find it, good luck =)

franck48, 2022-03-10:
Solution :
at the begining of the project i run CanOpenNode under FreeRtos taskS.
then, after remove it to keep code under 32k i put real-time process inside the same it-timer used by the stack.
but :
mr oscilloscope said :  i stay too long time inside it (sometimes), even with light treatment as it's mandatory to do.
the warning raised by CanOpenNOde was true....

from now, i just built a time base inside the it-timer,
all the code préviously inside the it is now in the main loop.

JuPrgn, 2022-03-10:
Great nice you found it ! 
Don't forget to close the issue (you could also change the title to be more explicit) =)

closed 2022-03-10 by franck48