CANopenNode/CANopenNode #382: Handling reception of RTR Frame for transmitting TPDO

NebulaDev16, 2022-03-09:
Hey,

We are looking into RTR frame for requesting the TPDO transmission. Transmission type of TPDO was 255.
So for this we made few quick changes around stack like passing rtr as 1 in CO_CANrxBufferInit() in CO_PDO.c
Still we could not get the CAN Stack transmitting PDO on reception on remote request
**can0  123   [0]  remote request**

We also  tried to playing around the transmission type of one PDO as 253 (Asynchronous RTR only). Still TPDO is not transmitted at reception of remote request. 

Please guide if we are doing something incorrect. Do we have this handling in CANOpen Node Stack?

NebulaDev16, 2022-03-09:
@martinwag quick response is highly appreciated

martinwag, 2022-03-09:
Sorry, can't help you there. Never tried to use RTR. I would assume it questionable if RTR is implemented anywhere in the stack. You would somehow need to receive it inside your driver, which I don't think any of the drivers implement RTR frames. You then would need a function/callback in the stack, which as far as I can remember doesn't exist.

aschaepper, 2022-03-10:
According to the documentation it is not?

"RTR (remote transmission request) is a feature of CAN bus. Usage of RTR is not recommended for CANopen and it is not implemented in CANopenNode."
From https://github.com/CANopenNode/CANopenNode und "some details"

Not sure if you want to implement it, I was just reading into this stack and saw this question...

Greeting

closed 2022-04-04 by NebulaDev16