CANopenNode/CANopenNode #383: Missing definitions when including CO.h

mathklk, 2022-03-18:
I used the most recent version of the CANopenEditor to export the files CO.h and CO.c. I am also using the most recent version of CANopenNode.

When compiling, the following errors get raised
```
lib\CanOpenNode\CANopen.c:58:3: error: #error OD_CNT_HB_PROD from OD.h not correct!
  #error OD_CNT_HB_PROD from OD.h not correct!
   ^
lib\CanOpenNode\CANopen.c:88:4: error: #error wrong OD_CNT_EM_PROD
   #error wrong OD_CNT_EM_PROD
    ^
lib\CanOpenNode\CANopen.c: In function 'CO_new':
lib\CanOpenNode\CANopen.c:277:51: error: 'CO_TX_CNT_EM_PROD' undeclared (first use in this function)
 #define CO_TX_IDX_TIME      (CO_TX_IDX_EM_PROD  + CO_TX_CNT_EM_PROD)
                                                   ^
...
(the compilation continues with subsequent errors, see full log below)
```

I searched the `OD.h` file and found no mention of these two defintions at all. They seem to be missing under "Counters of OD objects".

I assume there are some configurations missing in my profile, I'd greatly appreciate if anyone could help me find my mistakes.

Full compiler log:
[compiler_log.txt](https://github.com/CANopenNode/CANopenNode/files/8304375/compiler_log.txt)

My XDD-File:
[proFile_ECU2021_CO_OD.xdd.txt](https://github.com/CANopenNode/CANopenNode/files/8304378/proFile_ECU2021_CO_OD.xdd.txt)

Generated OD.h / OD.c files:
[OD.h.txt](https://github.com/CANopenNode/CANopenNode/files/8304381/OD.h.txt)
[OD.c.txt](https://github.com/CANopenNode/CANopenNode/files/8304382/OD.c.txt)

martinwag, 2022-03-18:
Those (at least some time ago) where linked to the existence of the corresponding OD entries (0x1017 for HB and I can't remember which one for EM). You can have a look at the .c/.h export methods in the editor source. Search for the lines that print the define.

CANopenNode, 2022-03-18:
You are probably using own EDS file and not xdd template file provided by CANopenNode. Note that there is no automation, when importing own EDS(or XDD) file. CANopenNode requires some custom properties to be set, especially in communication section. It is best to use original CANopenNode project file and add own objects. 

See 
https://github.com/CANopenNode/CANopenNode/tree/master#object-dictionary-editor
https://github.com/CANopenNode/CANopenNode/blob/master/doc/objectDictionary.md#object-dictionary-requirements-by-canopennode-object-dictionary-requirements-by-canopennode

closed 2023-03-28 by CANopenNode