CANopenNode/CANopenNode #384: Stable version

florenso, 2022-03-30:
I want a stable version to work with.

If I look at the tags, it seems that v2.0 is the newest tag. But there is also an v4.0. That confuses me.
https://github.com/CANopenNode/CANopenNode/tags

On the releases page it looks like V1.3 is the latest.
https://github.com/CANopenNode/CANopenNode/releases

what version is considered stable?

closed 2022-04-02 by florenso

vincent-chtln, 2024-12-04:
Hey @CANopenNode, do you intend to do a new tagged release (`4.1` or `5.0`) anytime soon ?

CANopenNode, 2024-12-04:
I could add a tag, but don't plan any special changes.

vincent-chtln, 2024-12-05:
Ok, so I understand it correctly, there's no "stable version" per say ? Only tagged commit for people needing to target a specific point in history ?

And you said in your previous comment:

> v4.0 is the beginning of the new design, but don't use that tag, just use the latest master, it is most stable. There is no release for it (yet).

Do you think the new design is polished enough now (or mostly stable) that you could consider making an "official" release ? "Official" might be a bit fancy but you see what I mean.
(That was what my first question was mainly about)

CANopenNode, 2024-12-05:
Yes, I can say, master is most polished and stable. So it is "official". I try to be conservative and not make unnecessary changes. There are polishing, fixes and some new features, that don't influence the rest code. Try to be careful. In case of larger change, a new tag or maybe a separate branch would be necessary.

Of course, I'm open for other opinions.