CANopenNode/CANopenNode #386: (CO_CANerror_rxMsgError) Socket error msg ID: 0x20000088

ytuemrah, 2022-04-07:
Hi All,
I have received the following messages, when I try to open can interface,
```
$ canopend can0 -i 1

canopend[11694]: CANopen device, Node ID = 0x00, starting
canopend[11694]: CAN Interface "can0" RX buffer set to 477 messages (212992 Bytes)
canopend[11694]: CANopen NMT state changed to: "initializing" (0)
canopend[11694]: CANopen device, Node ID = 0x01, communication reset
canopend[11694]: CANopen device, Node ID = 0x01, running ...
canopend[11694]: CANopen NMT state changed to: "pre-operational" (127)
canopend[11694]: (CO_CANerror_rxMsgError) Socket error msg ID: 0x20000088, Data[0..7]: 0x00, 0x00, 0x80, 0x19, 0x00, 0x00, 0x80, 0x00 (can0)
canopend[11694]: CANopen Emergency message from node 0x01: errorCode=0x5000, errorRegister=0x01, errorBit=0x2F, infoCode=0x00000004
canopend[11694]: (CO_CANerror_rxMsgError) Socket error msg ID: 0x20000088, Data[0..7]: 0x00, 0x00, 0x80, 0x19, 0x00, 0x00, 0x80, 0x00 (can0)
canopend[11694]: (CO_CANerror_rxMsgError) Socket error msg ID: 0x20000088, Data[0..7]: 0x00, 0x00, 0x80, 0x19, 0x00, 0x00, 0x80, 0x00 (can0)
canopend[11694]: (CO_CANerror_rxMsgError) Socket error msg ID: 0x20000088, Data[0..7]: 0x00, 0x00, 0x80, 0x19, 0x00, 0x00, 0x80, 0x00 (can0)
canopend[11694]: (CO_CANerror_rxMsgError) Socket error msg ID: 0x20000088, Data[0..7]: 0x00, 0x00, 0x80, 0x19, 0x00, 0x00, 0x80, 0x00 (can0)
canopend[11694]: (CO_CANerror_rxMsgError) Socket error msg ID: 0x20000088, Data[0..7]: 0x00, 0x00, 0x80, 0x19, 0x00, 0x00, 0x80, 0x00 (can0)
canopend[11694]: (CO_CANerror_rxMsgError) Socket error msg ID: 0x20000088, Data[0..7]: 0x00, 0x00, 0x80, 0x19, 0x00, 0x00, 0x80, 0x00 (can0)
canopend[11694]: (CO_CANerror_rxMsgError) Socket error msg ID: 0x20000088, Data[0..7]: 0x00, 0x00, 0x80, 0x19, 0x00, 0x00, 0x80, 0x00 (can0)
canopend[11694]: (CO_CANerror_rxMsgError) Socket error msg ID: 0x20000088, Data[0..7]: 0x00, 0x00, 0x80, 0x19, 0x00, 0x00, 0x80, 0x00 (can0)
canopend[11694]: (CO_CANerror_rxMsgError) Socket error msg ID: 0x20000088, Data[0..7]: 0x00, 0x00, 0x80, 0x19, 0x00, 0x00, 0x80, 0x00 (can0)
canopend[11694]: (CO_CANerror_rxMsgError) Socket error msg ID: 0x20000088...
```
I have arranged the interface as follows :
```
$ sudo ip link set can0 type can bitrate 125000
$ sudo ip link set up can0

$ ifconfig
can0: flags=193<UP,RUNNING,NOARP>  mtu 16
        unspec 00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00  txqueuelen 10  (UNSPEC)
        RX packets 0  bytes 0 (0.0 B)
        RX errors 5385343  dropped 71  overruns 0  frame 0
        TX packets 13  bytes 62 (62.0 B)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0
```
the candump output is as follows :
```
$ candump -td -a can0

 (000.000000)  can0  701   [1]  00                        '.'
 (000.000734)  can0  081   [8]  00 50 01 2F 04 00 00 00   '.P./....'
```
I am a user of ubuntu as follows :
```
$ uname -a
Linux ubuntu 5.13.0-39-generic #44~20.04.1-Ubuntu SMP Thu Mar 24 16:43:35 UTC 2022 x86_64 x86_64 x86_64 GNU/Linux
```
I am user of PCAN-USB :
```
ubuntu@ubuntu:~/Desktop/CANopenLinux$ lsmod |grep can
can_raw                20480  0
can                    20480  1 can_raw
pcan                  188416  0
can_dev                36864  1 pcan
i2c_algo_bit           16384  3 pcan,i915,nouveau
```
I have followed the steps as you have mentioned in your linux document :

> USB, PCI or similar CAN interface
> There are several CAN interfaces on the market which works with Linux SocketCAN. See [Linux kernel source](https://git.kernel.org/cgit/linux/kernel/git/torvalds/linux.git/tree/drivers/net/can), Kconfig files, for supported interfaces by the Linux kernel. For example [EMS CPC-USB](https://www.ems-wuensche.com/?post_type=product&p=746) or [PCAN-USB FD](http://www.peak-system.com/PCAN-USB-FD.365.0.html?&L=1). Usually such interface is started with:
> 
> sudo ip link set up can0 type can bitrate 250000

I have searched about it but I have not reached any solution, could you guide me? Do you have solution about it?
Best Regards

martinwag, 2022-04-07:
include/linux/can/error.h of your kernel sources include the info needed to decode "Socket error msg ID" messages. You need to understand CAN hardware and the kernel driver implementation to your CAN interface to make sense of those.

ytuemrah, 2022-04-14:
Dear Martin,
Thank you for your feedback, as you sad it is about kernel driver and pcan-usb hardware. Also, the peak-system supports at that point.
Best Regards

closed 2022-04-14 by ytuemrah