CANopenNode/CANopenNode #388: Guidance in testing C2000 support

sicrisembay, 2022-04-16:
Hi,
I am currently trying to port support for C2000 (I'm using TMS320F28335).
[CANopenNode_C2000](https://github.com/sicrisembay/CANopenNode_C2000.git)

This support/port is quite messy (I don't see this thing getting PR'd in the near future).
16-bit wide byte makes it difficult but I made it work (so far tested working is SDO).

I am not familiar on CANopen conformance test.  I need guidance on how to test my C2000 port.   I hope that somebody has done this and can share his/her experience.  Thanks in advance.

best regards,
Sicris

sicrisembay, 2022-04-16:
While exploring @CANopenNode repos, I found [CANopenDemo](https://github.com/CANopenNode/CANopenDemo.git) as a good starting point for testing.  Am I on the correct path?

martinwag, 2022-04-19:
If you're CiA member (or intend to become one), there is a conformance test tool.

sicrisembay, 2022-04-20:
I am not CiA member and I am just doing this out of interest and on my free time (just got my hands on TI HvMtrPfcKit which is lying around).  I see some test cases written using BATS in the CANopenDemo.  I think that it's a good starting point for me.

If somebody who is interested running CANopenNode on C2000 and is a CiA member.  I am willing to spend some time to improve C2000 port.  As for now, I'll just test whatever I can.

martinwag, 2022-04-20:
In that case I would recommend you to get one of the many ARM devboards (ST, NXP, ...). They are ~15€/$ and you will learn something more useful than fiddling with obscure architecture :-)

sicrisembay, 2022-04-20:
I have both ST (stm32f4xx) and NXP (imxrt1052-evk) devkits.  I did run CANopenNode on that kits few years back and I am very happy with CANopenNode.  :)

I haven't seen open-source CANopen stack for C2000.  I've used commercial CANopen stack on TM320F28335 when I was working with Wind Turbine company  ~10 yrs ago.  Out of curiosity, I am interested on idea on adding support for obscure C2000 DSP architecture.  I still have time up to this weekend to fiddle then I'll move into something else.

So far, I've done a quick test on:
EMCY
NMT
SDO
SYNC

TODO this weekend
PDO
LSS
... and fix porting issues found along the way.

sicrisembay, 2022-05-01:
I finally able to test PDO's (RPDO and TPDO) on C2000.
I already satisfied my curiosity and time to move on to something else.

Closing...

closed 2022-05-01 by sicrisembay

AnthoJack, 2023-04-20:
Hi,
I would be interested in your code if you still have it. I'm working on the TMS320F28388D and would love to have some (even partially) working code to use as a basis for my driver

sicrisembay, 2023-04-21:
@AnthoJack  [CANopenNode_C2000](https://github.com/sicrisembay/CANopenNode_C2000) is a working code.  Adapting it for a more advance C2000 MCU should not be difficult.  Your MCU has CAN-FD peripheral.  I suggest that you consider CANopen FD