CANopenNode/CANopenNode #390: Transmit and Receive PDO

IgorKaan, 2022-04-19:
Hi! I'm new at CANOpen. Trying to communicate 2 STM32F103RB Nucleo boards.  I have taken CANopenNode/CanOpenSTM32/tree/master/stm32f0xx_can example and adapted it to my MCU, code started without errors (diagnostic messages has no errors). Now how can i transmit and receive PDO objects, which functions i should use? I can see CO_process, CO_process_SYNC, CO_process_TPDO and CO_process_RPDO called every 1 ms in example.

martinwag, 2022-04-20:
You map your OD entries to PDOs. Then a value change will trigger PDO transmission. No manual sending needed.

HamedJafarzadeh, 2022-04-20:
> 

Martin, I believe that's not the case for the v4. You have to explicitly call  `CO_TPDOsendRequest(&CO->TPDO[0]);` to send a PDO. 

@IgorKaan , I would suggest monitoring your CAN bus for startup message from nodes, plus configuring the heartbeat time to some values so that you can see regular messages on the CAN bus. Then you can proceed to the next step and generate Object Dictionary and PDO's and mappings. Make sure you use EDSGUI (CanOpenNodeEditor), as I realized the OD file cannot be modified with other tools. I did couple of fixes to EDSGUI where you can find it (here)[https://github.com/HamedJafarzadeh/CANopenEditor/releases/tag/v4.0.1-a]. It is not a production-ready build anyway, but stable enough for my use cases.

CANopenNode, 2022-04-20:
> Martin, I believe that's not the case for the v4. You have to explicitly call `CO_TPDOsendRequest(&CO->TPDO[0]);` to send a PDO.

I have to clarify this a little. TPDO have multiple triggers in all versions, some are standard and some are application specific. Most simple is event timer, which is standard. So, you just map your (mappable) OD entries to PDOs, set event timer, enable TPDO and you should see them in candump. See https://canopennode.github.io/CANopenNode/group__CO__PDO.html#details

There is a difference in application specific TPDO triggers between <=V2 and V4. (Versions before V4 have an old object dictionary interface.):
1. In old versions there was a checkbox "COS trigger" in Object dictionary editor (libedssharp, now CanOpenEditor). So, if OD entry has this "ChangeOfState" flag set then, TPDO is triggered automatically by the stack on change of state of the variable.
2. In V4 this is made more flexible: application may verify change of state of some variable (ore some other condition) and call [OD_requestTPDO()](https://canopennode.github.io/CANopenNode/group__CO__ODinterface.html#gac3164283191a778559f10e331bc9d775). Stack will then send zero or more TPDOs to which OD variable is mapped, according to rules specified by TPDO transmission types. You can also use [CO_TPDOsendRequest()](https://canopennode.github.io/CANopenNode/group__CO__PDO.html#ga83b48c441965596e6d693f879c076ae4) to trigger TPDO directly, but with this you lose flexibility of PDO mappings.

IgorKaan, 2022-04-20:
> Martin, I believe that's not the case for the v4. You have to explicitly call `CO_TPDOsendRequest(&CO->TPDO[0]);` to send a PDO.
>
> @IgorKaan , I would suggest monitoring your CAN bus for startup [...]
@[HamedJafarzadeh](https://github.com/HamedJafarzadeh) Hi! I can't see any startup messages from node with candump, when i start CANopenNode code on one of MCU, but im sure that CAN connection is OK, i can send and receive data with HAL_CAN_AddTxMessage and HAL_CAN_GetRxMessage HAL functions

HamedJafarzadeh, 2022-04-20:
> 

@IgorKaan Just cloned CANOpenSTM32, built, programmed on my NUCLEO board and I have it working on my desk right now 😄. Make sure of your hardware, board and MCU that you are using. I assumed that you are using NUCLEO boards. Have a look at the [CANOpenSTM32 - Porting tips](https://github.com/CANopenNode/CanOpenSTM32#when-usingporting-nucleo-examples-do-not-forget-to-). Use breakpoints in `prv_send_can_message` and `CO_CANinterrupt_TX` functions to see if your program make it to these lines. It shouldn't be very hard to debug it and find the issue I would say. I'm using PCAN for monitoring the bus so I'm not quite familiar with candump unfortunately, being said that, make sure of configuration on that side ( rx filters and etc ).

IgorKaan, 2022-04-20:
> > 
> 
> @IgorKaan Just cloned CANOpenSTM32, built, programmed on my NUCLEO board and I have it working on my desk right now smile. Make sure of your hardware, board and MCU that you are using. I assumed that you are using NUCLEO boards. Have a look at the [CANOpenSTM32 - Porting tips](https://github.com/CANopenNode/CanOpenSTM32#when-usingporting-nucleo-examples-do-not-forget-to-). Use breakpoints in `prv_send_can_message` and `CO_CANinterrupt_TX` functions to see if your program make it to these lines. It shouldn't be very hard to debug it and find the issue I would say. I'm using PCAN for monitoring the bus so I'm not quite familiar with candump unfortunately, being said that, make sure of configuration on that side ( rx filters and etc ).

IgorKaan, 2022-04-20:
> @IgorKaan Just cloned CANOpenSTM32, built, programmed on my NUCLEO board and I have it working on my desk right now smile. Make sure of your hardware, board and MCU that you are using. I assumed [...]

@[HamedJafarzadeh](https://github.com/HamedJafarzadeh) Now have just made a little fixes and now i can see five msgs with STdID 0x70C = 0x700 + node id and nothing more. Is it OK?

HamedJafarzadeh, 2022-04-20:
> 

@IgorKaan yes, that's correct. It means that the transmitting part is fine. You can now try to read SDOs like device type from master side and see if the node responds back. if it doesn't respond have a look at `CO_CANInterrupt_TX` and CAN interrupts in general.

IgorKaan, 2022-04-20:
Finally it works! I can control my MCU  with SDO and transmit data from sensors with PDO, great

IgorKaan, 2022-04-20:
[CANopenNode](https://github.com/CANopenNode) 
[HamedJafarzadeh](https://github.com/HamedJafarzadeh)

Now i nave another problem, 
As described here:
https://github.com/CANopenNode/CANopenNode/blob/master/doc/objectDictionary.md

"Some simple user applications can also access some OD variables directly via globals.
@warning If OD object has OD extension enabled, then direct access to its OD variables must not be used. Only valid access is via read or write or helper functions."

I'm trying to transmit variables, like 
OD_RAM.x6401_readAnalogInput_16_bit[0] = OD_value_1;
OD_RAM.x6401_readAnalogInput_16_bit[1] = OD_value_2;
And it works.
When im trying to transmit
OD_RAM.x6401_readAnalogInput_16_bit[0] = OD_value_1;
OD_RAM.x6401_readAnalogInput_16_bit[1] = OD_value_2;
OD_RAM.x6401_readAnalogInput_16_bit[2] = OD_value_3;
It doesn't work and red diode flashing..
As i see OD.c 
    .x1A01_TPDOMappingParameter = {
        .numberOfMappedApplicationObjectsInPDO = 0x04,
        .applicationObject_1 = 0x64010110,
        .applicationObject_2 = 0x64010210,
        .applicationObject_3 = 0x64010310,
        .applicationObject_4 = 0x64010410,
        .applicationObject_5 = 0x00000000,
        .applicationObject_6 = 0x00000000,
        .applicationObject_7 = 0x00000000,
        .applicationObject_8 = 0x00000000
    }
mapped for 4 elements, and i think three variables should be transmitted without problems. What's going wrong?

CANopenNode, 2022-04-21:
> ... red diode flashing..

Is there any emergency message?

IgorKaan, 2022-04-21:
@[CANopenNode](https://github.com/CANopenNode)
Sorry, red diode is blinking 2,5 Hz and emergency message from MCU node
Is that correct way to access OD via global variables? 
when i'm starting with code
if(htim == &htim2) {
  //OD_RAM.x2110_variableInt32[0] = OD_value32_1;
  OD_RAM.x6401_readAnalogInput_16_bit[0] = OD_value_;
  OD_RAM.x6401_readAnalogInput_16_bit[1] = OD_value_1;
  OD_RAM.x6401_readAnalogInput_16_bit[2] = OD_value_2;
  OD_RAM.x6401_readAnalogInput_16_bit[3] = OD_value_3;
}
it's OK, and i receive OD_RAM.x2110_variableInt32[0] with zero value, but when i start 
if(htim == &htim2) {
  OD_RAM.x2110_variableInt32[0] = OD_value32_1;
  OD_RAM.x6401_readAnalogInput_16_bit[0] = OD_value_;
  OD_RAM.x6401_readAnalogInput_16_bit[1] = OD_value_1;
  OD_RAM.x6401_readAnalogInput_16_bit[2] = OD_value_2;
  OD_RAM.x6401_readAnalogInput_16_bit[3] = OD_value_3;
}
red diode is blinking, i receive data 1 time, and emergency message from node... 
 CANopen Emergency message from node 0x09: errorCode=0x8110, errorRegister=0x10, errorBit=0x14, infoCode=0x00000000

closed 2022-04-21 by IgorKaan

CANopenNode, 2022-04-22:
Please try to understand the emergency message:
https://canopennode.github.io/CANopenNode/group__CO__Emergency.html
Then experiment with small steps.